witnessmenow/Universal-Arduino-Telegram-Bot #144: WebServer and telegram together

array81, 2020-02-13:
On my project I use a webserver and telegram.
Today, after some months, I have restart my device and I have get problem with SSL connection to telegram.
So have upgrade to last versione ESP8266 library to solve SSL problem. I have remove HTTPSRedirect and now I use WiFiClientSecure with Fingerprint. Telegram works until visit webserver page. If I try to visit the webserver page after some second I get for some second this error:
```
GET Update Messages
[BOT]Connecting to server
[BOT]Conection error
```
and then the device restart with this error:
```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000df64
EXCVADDR: 0x00000000

Decoding stack results
0x40100f9c: calloc(size_t, size_t) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 820
0x4022b1ac: tcpip_tcp_timer at core/timeouts.c line 145
0x40100d44: malloc(size_t) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 511
0x40217de7: loop_task(ETSEvent*) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 188
0x40236208: br_sha384_update at src/hash/sha2big.c line 165
0x40236a8f: sha2small_out at src/hash/sha2small.c line 249
0x40245051: br_multihash_update at src/hash/multihash.c line 141
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x4022915d: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x40217e97: optimistic_yield(uint32_t) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 129
0x40100a84: umm_free_core(void*) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 316
0x40100d0f: free(void*) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40100d0f: free(void*) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 160
0x40217e67: __yield() at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 100
0x40217ea5: optimistic_yield(uint32_t) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 133
0x4020a24f: WiFiClient::available() at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\WiFiClient.cpp line 263
0x4020bed6: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 527
0x4023879e: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1305
0x4020c138: BearSSL::WiFiClientSecure::_wait_for_handshake() at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 564
0x4020c31b: BearSSL::WiFiClientSecure::_connectSSL(char const*) at c:\users\martin\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\bits/shared_ptr.h line 291
0x40218656: __delay(unsigned long) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_wiring.cpp line 54
0x4020add4: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src/include/ClientContext.h line 133
0x4020c439: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 232
0x40214cf0: UniversalTelegramBot::sendPostToTelegram(String, ArduinoJson::JsonObject&) at C:\Users\Martin\Documents\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp line 104
0x40216c93: String::reserve(unsigned int) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 146
0x40216cdc: String::copy(char const*, unsigned int) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 214
0x40216ec0: String::operator=(String const&) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 262
0x40216ef0: String::String(String const&) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 41
0x40214f48: UniversalTelegramBot::sendPostMessage(ArduinoJson::JsonObject&) at C:\Users\Martin\Documents\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp line 621
0x40221400: ESP8266WiFiScanClass::RSSI(unsigned char) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\ESP8266WiFiScan.cpp line 232
0x40213ed5: ArduinoJson::JsonObject::set (char const*, String const&) at C:\Users\Martin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../StringTraits/StdString.hpp line 28
0x40215021: UniversalTelegramBot::sendMessage(String, String, String) at C:\Users\Martin\Documents\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp line 540
0x40216cdc: String::copy(char const*, unsigned int) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 214
0x40201561: Log(int, String) at C:\Users\Martin\OneDrive\Programmazione\mrGarden2/mrGarden2.ino line 282
0x40216cdc: String::copy(char const*, unsigned int) at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 214
0x40207d2d: setup() at C:\Users\Martin\OneDrive\Programmazione\mrGarden2/mrGarden2.ino line 1537
0x40221444: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\martin\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40217f48: loop_wrapper() at C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 177
```
How can I solve it?

I use a NodeMCU.

qlalfdu, 2020-02-26:
have a similar problem.
There was no problem for 30 hours, sometimes with an explosive reset every 1-3 minutes.
I checked the ping and there are many times when there is no response for more than 5 seconds.

witnessmenow, 2020-09-30:
Not sure there is much I can do about this, so not a library issue.

closed 2020-09-30 by witnessmenow