witnessmenow/Universal-Arduino-Telegram-Bot #147: Stops after 2 days

fmarzocca, 2020-02-21:
I have an esp8266 installed as an unmanned warning system, but afetr 2 days it stopped receiving messages from the bot. Other functions in the device are still workign (i.e. webserver), but it is not receiving any messages from the bot. I had to restart the device. If this is the normal behaviour, it is not reliable for a warning system.

This is my  loop function (Bot_mtbs is 1000):

```
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    while (numNewMessages) {
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }
    Bot_lasttime = millis();
```

EcstasyAwesome, 2020-02-21:
I have the same problem. Some time ago everything worked fine.
Esp8266 lib v2.6.3

burdeyniyua, 2020-02-21:
Скорее всего менялось api телеграмма.
Поставил версию ядра esp 2.5.0
А в setup добавил
client.setInsecure();

EcstasyAwesome, 2020-02-21:
> Скорее всего менялось api телеграмма.
> Поставил версию ядра esp 2.5.0
> А в setup добавил
> client.setInsecure();

Они должны сохранять обратную совместимость. По этому не соглашусь по поводу api.
А на счет esp возможно, эта проблема появилась после перехода на версию 2.6.3. До этого была 2.4.2 и все стабильно работало.

burdeyniyua, 2020-02-21:
У меня бот перестал отвечать сам, без перехода на обновленную версию прошивки((
Поэтому тут же на форуме нашел такое решение.
Да должна быть обратная совместимость.
Перезагрузка можно прописать, но у меня на последней версии отказывается работать (

EcstasyAwesome, 2020-02-21:
```
#define ONLINE_TIME 86400000 // 24 hours

void loop() {
  if (millis() > ONLINE_TIME) ESP.restart();
}
```
 Такой код работает корректно на версии 2.6.3

@burdeyniyua на версии 2.5.0 нет проблемы, которая здесь описывается?

fmarzocca, 2020-02-21:
Can you pls write in english? Otherwise I will start writing in italian...

fmarzocca, 2020-02-21:
I am using 2.6.3 in board manager.

EcstasyAwesome, 2020-02-21:
Downgrade your version of esp to 2.5.0 and watching for a few days. Report your result.
Or try to use my code above.

fmarzocca, 2020-02-21:
> Downgrade your version of esp to 2.5.0 and watching for a few days. Report your result.
> Or try to use my code above.

Thank for answering. I will see what to do. The device is for a warning system so I will be reluctant to restart it every day...

EcstasyAwesome, 2020-02-21:
[the difference between ESP.reset() and ESP.restart()](https://www.pieterverhees.nl/sparklesagarbage/esp8266/130-difference-between-esp-reset-and-esp-restart)

fmarzocca, 2020-02-21:
Thanks! It seems that ESP.restart() is more safe than ESp.reset()

- Sent from my mobile phone. Sorry for typos -

fmarzocca, 2020-02-24:
@EcstasyAwesome , do you know why with certain APs (RSSI: 58%), the device response is very slow (almost 1 minute from the telegram command)?

EcstasyAwesome, 2020-02-24:
@fmarzocca, my current device is working stable with the signal less than yours. Maybe the problem is in a code or an access point. I can say you no more without your code.
Did you resolve the previous problem?

fmarzocca, 2020-02-24:
I have installed the sketch with ESP FW 2.5.0, and it is still working after 2 days. Let's see incoming days.
But I have this weird problem with an AP  (it is an extender). I need a signal >75% to have telegramBot working. With lower signals other functionas are still working (i.e.: webserver, OTA) but not telegram.
My telegram code is quite straightforward (it is in my first post here).

EcstasyAwesome, 2020-02-24:
Try to reconfigure a wifi AP and a repeater with the following settings:
- mode 2.4GHz (better stability at long ranges as I know)
- experiment with the channel selection or use "Auto"
- set the channel width to 20 MHz (the performance tradeoffs from interference)

fmarzocca, 2020-02-24:
Got it. The 20mhz settings is much more stable, thanks. I will continue testing on the other issue (FW 2.5.0) and report here.

fmarzocca, 2020-02-24:
When I send a command, the mean time for the response is 10sec. Is it normal? Can it be improved?

EcstasyAwesome, 2020-02-24:
As I know with the FW 2.5.2 works fine too.
My ESP8266 receives a message from a bot, does request to an Arduino and after gets some data (via serial port) and sends it to me. It has taken a few seconds. Draw your conclusions.

fmarzocca, 2020-02-24:
I moved to 2.52. Will report results.
I have also tested several ESP8266, and there are differences on WiFi perfomances. These chinese chips are not so much reliable...

EcstasyAwesome, 2020-02-24:
I have tested only one and he has been stable working since the previous winter. But everything can be...

fmarzocca, 2020-02-24:
I switched back to 2.6.3. It is very much faster than 2.5.x! I will then implement your solution to reset the device every 24h.

javierferwolf, 2020-02-25:
@EcstasyAwesome @burdeyniyua привет ребята, у меня есть такая проблема и из того, что я прочитал здесь! я решил переходить на версию 2.4.2! но уже не рабочая версия оказывается, я в setup добавил client.setInsecure(); но не компилируется получаю ошибку axTLS::WiFiClientSecure has no member named 'setInescure'!! потом меняю WiFiClientSecure client; на  BearSSL::WiFiClientSecure client; теперь вроде код компилируется! но телеграмм еще не работает! есть какой-нибудь решение или с этой проблемой столкнулись?

fmarzocca, 2020-02-25:
Hi  Vladyslav, I made a packet capture on my device and I have discovered that there is a very high activity every few mSeconds! I sthere a way to reduce this activity, but in the same time keep a reasonable response time?

EcstasyAwesome, 2020-02-25:
@fmarzocca, hi! What kind of activity do you talk about?
@javierferwolf FW 2.4.2 с недавних пор не работает с телегой. Используй FW 2.5.0 и выше и WiFiClientSecure client. В setup() добавляй client.setInsecure(); и будет работать.

fmarzocca, 2020-02-25:
Attached is just a short screenshot. You can see that activity is every few milliseconds, without typing anything in Telegram.. 192.168.1.65 is the device's local IP.

[image]

fmarzocca, 2020-02-25:
and these are the relevant lines from my sketch concerning Telegram Bot:

```

#define BOTtoken "xxxxxxxx"
#define CHAN_ID "xxxxxxxxx"
WiFiClientSecure clientSecure;
UniversalTelegramBot bot(BOTtoken, clientSecure);

int Bot_mtbs = 1500; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

void setup() {
	....
	clientSecure.setInsecure();
	....
	}

void loop() {
	....
	if (millis() > Bot_lasttime + Bot_mtbs)  {
		int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
		while (numNewMessages) {
			handleNewMessages(numNewMessages);
			numNewMessages = bot.getUpdates(bot.last_message_received + 1);
		}
		Bot_lasttime = millis();
	}
}
```

EcstasyAwesome, 2020-02-25:
This 149.154.167.220 address is used by Telegram.
Maybe, the telegram bot library works like that. Anyway, you should check the library code to find the answer.

fmarzocca, 2020-02-25:
Yes, it works that way. But it is not acceptable... I will make my decision on this project.

javierferwolf, 2020-02-25:
@EcstasyAwesome спасибо за быстрый ответ,  я конечно поставил 2.5.0 версию но почему-то перестает работать другой функции программы. У меня сначала стоял 2.6.3 версия но у меня были две проблемы одна что телег перестал работать через день или два дня, и другая проблема что в версии 2.6.3 почему-то не работала система архива правильно SPIFFS. и вдруг тут нашел решение первой проблемой, решил переходить в версию 2.4.2 но телеграмм не работает НО система -архива SPIFFS работает нормально.

EcstasyAwesome, 2020-02-25:
@javierferwolf, может [это](https://github.com/esp8266/Arduino/issues/7095) поможет с второй проблемой.
Используя версию 2.6.3 нужно делать перезагрузку через определенное время, я выше об этом писал. Может эту проблему можно решит по другому, но я использую такой вариант.
В версии 2.5.0 такой проблемы нет (в 2.5.2 вроде тоже, но утверждать не буду)

fmarzocca, 2020-02-25:
@javierferwolf , when you switch to core >=2.6.2, you need to rebuild the Arduino FS plugin. See: https://github.com/esp8266/Arduino/pull/6807

javierferwolf, 2020-02-25:
@fmarzocca thanks for your advice! my situation is a bit strange, first of all I have always used version 2.6.3 and the spiffs file system works very well in other project codes that I have used. but now I want to automate my old home alarm system with telegram and with an IP camera, which basically detects a movement, the camera makes an instant and is saved in SPIFFS and then sends me via telegram.  The first problem was that the ESP8266 did not save the photo in spiffs because the ESP was rested at that moment, it had an Exception (0) error:
epc1 = 0x40212d48 epc2 = 0x00000000 epc3 = 0x00000000 excvaddr = 0x00000000 depc = 0x00000000.  
Then delete the option of the IP camera and the photos! and then I just made a program code that just let me know the alarm by telegram. the program was working fine for two days or three then it freezes. the alarm code works fine but telegram did not send the messages I have to turn it off and on to make it work again.  and today I decided to change to version 2.4.2 and to my surprise the part of saving the photo in SPIFFS works but now telegram does not work.

javierferwolf, 2020-02-25:
@fmarzocca but just in case how can I update the plugin?:)

fmarzocca, 2020-02-25:
Just  reinstall del FS plugin, but I believe it is not your problem. Maybe the SPIFFS is not large enough for the picture or your saving routines are buggy? I don't know how to help you here.

COncerning your other problem, as Vladyslav suggested, you can run 2.6.3 core (which is faster) and put this in the loop:

```
#define ONLINE_TIME 86400000 // 24 hours

void loop() {
  if (millis() > ONLINE_TIME) ESP.restart();
}
```

The device will restart every 24 hours (just few seconds), so your telegram bt can continue working.

tadcrazio, 2022-12-29:
FWIW - I'm a couple years late here, but alas i was having this issue. After 2 days the bot would stop functioning. I switched to a different telegram library and i haven't had any issues weeks in. I suspect there to be a bug somewhere causing this.

AvalancheJoseki, 2023-01-14:
@tadcrazio, what library did you move from and to?