witnessmenow/Universal-Arduino-Telegram-Bot #148: Abnormal huge network activity

fmarzocca, 2020-02-26:
I have just performed an ethernet capture over my ESP8266 using this library, and these are the results!
[image]

There is an huge network activity, packets sent and received many times/second towards/from api.telegram.org. 
These traffic is generated by the most simple sketch and without interacting with Telegram, just waiting. 

```
#define BOTtoken "xxxxxxxx"
#define CHAN_ID "xxxxxxxxx"
WiFiClientSecure clientSecure;
UniversalTelegramBot bot(BOTtoken, clientSecure);

int Bot_mtbs = 1500; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

void setup() {
	....
	clientSecure.setInsecure();
	....
	}

void loop() {
	....
	if (millis() > Bot_lasttime + Bot_mtbs)  {
		int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
		while (numNewMessages) {
			handleNewMessages(numNewMessages);
			numNewMessages = bot.getUpdates(bot.last_message_received + 1);
		}
		Bot_lasttime = millis();
	}
}

```

If this is normal, I will not use this library and suggest not to use, as it overchage your AP and the ESP cpu.
I hope there is a way to bring this traffic to normal. I have tested python libraries on Raspberry and there are very "silent".

denzen84, 2020-02-26:
try to use at setup()

`bot.longPoll = 50;`

fmarzocca, 2020-02-26:
> try to use at setup()
> 
> `bot.longPoll = 50;`

50 seconds of long poll?? This will block my device for a minute. I am using OTA to update it and it includes also a webserver to listen to http requests. I can't.
But the activity is generated only towards telegram.org continuosly. I have a mtbs= 1500 mseconds, but it does not comply with it.

denzen84, 2020-02-26:
You're wrong. It means that telegram server will keep connection for 50 seconds with your device after SSL handshake. If you send command or receive the response, telegram server will send it immediately.

Default setting for long pool is 0 (as it used in universal telegram bot). It this case each controller every second (sic!) creates SSL connection and receives NULL result. And does it again and again. Average traffic is 16kb per each SSL handshake. Every controller with universal telegram bot produces about 30-40Gbytes of junk traffic montly. Setting maximal value of 50s for longPool reduces this junk traffic to 3Gbytes per month and have no affect for time of reaction. Just trust me, test your case with longPool = 50 and you will dramatically reduce traffic without any additional loses.

fmarzocca, 2020-02-26:
I could test it, but I remember I set it to 60 few weeks ago, and the OTA didn't work anymore as the loop was delayed.
This is my loop:

```
void loop() {
	server.handleClient();
	ArduinoOTA.handle();
	timeClient.update();
        if (millis() > Bot_lasttime + Bot_mtbs)  {
		int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
		while (numNewMessages) {
			handleNewMessages(numNewMessages);
			numNewMessages = bot.getUpdates(bot.last_message_received + 1);
		}
		Bot_lasttime = millis();
	}
}
```
If I set longPoll=50, the `ArduinoOTA.handle()` and `server.handleClient()`will be blocked for 50 seconds.

denzen84, 2020-02-27:
As I understand, you found a solution to use LongPoll and async TCP connection.

fmarzocca, 2020-02-27:
> As I understand, you found a solution to use LongPoll and async TCP connection.

Not really async TCP, but the mods introduced by RomanLut are good for my purpose. A non-blocking longPoll.

closed 2020-02-27 by fmarzocca