witnessmenow/Universal-Arduino-Telegram-Bot #153: Update BotToken during Setup phase

rafaelcbarbosa, 2020-03-29:
In all examples, I can see that the bot is being declared as a global variable. This requires the BOTtoken to be available very early in the boot sequence. 

I am storing the BOTtoken in EEPROM and would like to have the bot defined during setup, while keeping its global scope.

Is it possible? (let me know if you need the code)

witnessmenow, 2020-03-29:
Hello,

This example does what you describe:

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/UsingWiFiManager/UsingWiFiManager.ino

Hope this helps,
Brian

witnessmenow, 2020-03-29:
There is WiFi manger stuff you might not need, but it shows reading the
token from eeprom and using it in the setup

rafaelcbarbosa, 2020-03-29:
Exactly what I needed. 

I was missing the `new` for the declaration inside the setup. Thank you very much!!!!

If you wish I can send you my example, I am using Wifi Server for 1-time setup (as per your example), interruption (to capture the prg button and enter in setup mode), web server for the page to capture info, EEPROM to save the data. Very similar, I wish I have found this ~1 year ago :)

closed 2020-03-29 by rafaelcbarbosa

ResearchEspl, 2021-01-14:
> Hello, This example does what you describe: https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/UsingWiFiManager/UsingWiFiManager.ino Hope this helps, Brian
> […](#)

Link provided as https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/UsingWiFiManager/UsingWiFiManager.ino is not opening. Can you please send me code in which UniversalTelegramBot is declared in global and new token read from eeprom in setup and again new token read from eeprom is updated.

Kinder1664, 2025-01-02:
> > Здравствуйте, Этот пример делает то, что вы описываете: https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/UsingWiFiManager/UsingWiFiManager.ino Надеюсь, это поможет, Брайан 
> > […](#) 
> 
> Ссылка предоставлена ​​как https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/UsingWiFiManager/UsingWiFiManager.ino не открывается. Можете ли вы прислать мне код, в котором UniversalTelegramBot объявлен в глобальном масштабе и новый токен считывается из eeprom в настройке и снова новый токен считывается из eeprom обновляется.

Hello. Could you share it with me too. I can't use the token I read from the EEPROM.