witnessmenow/Universal-Arduino-Telegram-Bot #163: esp8266 examles are not woring anymore

555learnwithme555, 2020-04-25:
non of the examples are working even after downgrading the board version to 2.5.2 and also tried adding client.setInsecure()......... hoping someone would help me finding a solution...

MaitrePo, 2020-04-25:
Where did you add client.setInsecure() ?

555learnwithme555, 2020-04-28:
i added it in void setup

MaitrePo, 2020-04-28:
Which example did you try ? I added client.setInsecure() at the end of the setup() method.
I added some logs with Serial.println() in my program.
I also set to "true" the "debug" variable in UniversalTelegramBot.h (in Arduino\libraries\Universal-Arduino-Telegram-Bot-master\src) .
Maybe this would help you to be sure that there is not any other problem ?

ShrileshKale, 2020-05-01:
@MaitrePo 
I did what you said above but no success. 

- board version -2.5.2

- Code - Picked up from sample code (e.g Flash led)

- Added client.setInsecure() in void.setup()

[image]

I am not getting any output after this..Could you please help me?

prontodos, 2020-05-02:
Same for me but it starting working well after I added this line below the void setup() 

client.setInsecure();

ShrileshKale, 2020-05-02:
Ok.what is your Arduino IDE and ESP8266 board version?

prontodos, 2020-05-02:
esp8266 (now) 2.7.0
2.6.3 worked as well but I read here that 2.5.x are are stable. 

arduino ide 1.8.12

good luck

ShrileshKale, 2020-05-02:
@prontodos No success with 2.7 , 2.6.3. What to do now?

sicamxdf, 2020-07-05:
the examples for flash led does not work always have the same response {"ok":false,"error_code":401,"description":"Unauthorized"}

esp8266 (now) 2.7.0
arduino ide 1.8.12
thanks in advance

witnessmenow, 2020-09-30:
Examples have all been updated on master branch. Will be released soon, but for now use the Github version of the library.

closed 2020-09-30 by witnessmenow