witnessmenow/Universal-Arduino-Telegram-Bot #166: 1 esp32 to multiple telegram bots

prontodos, 2020-04-29:
I tried echobot on my esp32 to a telegram bot that I just set up and it worked well. 
I then tried the same echobot on the same esp32 to ANOTHER telegram bot (my existing doorbell bot) 
and it wouldn't work. 

Can you pls tell me if this is possible ?

And also if I were to use 1 esp32 to send to multiple telegram bots will this work ?

witnessmenow, 2020-04-29:
I think you should be able to do this, not sure if would preform very well
but I think it's possible.

Is the second bot still on your doorbell project? Cause the other project
might "deal" with the message before the esp32 gets a chance to see it

Also use the same client on both bots

prontodos, 2020-04-29:
Thank you for you reply. 
here is what I have tried:

using echobot on my esp32 with newly create bot works. 

when using the same esp32 echobot but changed sketch bot ID to my doorbell bot it will not work.
nothing is echoed. 

Yes the doorbell bot is still alive and doing it's work.
My doorbell bot just sends me a picture when the doorbell button is pressed.

I was hoping both my echo bot esp32 AND doorbell can share the same doorbell bot. 

I idea is to have a few esp32s working at the same time and sending info to the same tele bot.

witnessmenow, 2020-04-29:
The way telegram bots makes it very difficult to have same bot on multiple devices.

When you send a message to your bot, it sits on Telegrams servers waiting for your code to come get the message. The default behavior of most libraries, including this one, basically deletes the message from Telegrams server once it has successfully read the message.

it does this by the ID that is passed in on getUpdates, you'll see in this libraries examples it's the ID of the last message you read + 1. This basically tells Telegram that you have already dealt with any messages with a lower number than this. This is necessary though, as if you dont tell Telegram you have dealt with that message, it will keep sending it back to you.

So if you have two devices with the same bot token, device A could read the message and mark it for deletion before device B sees it. I'm assuming your doorbell is a Raspberry pi or something? That would be able to check for messages much faster than an ESP32 so it would nearly always deal with the message first.

There is no issue with having multiple devices send from the one bot token, the issue is only with reading. It may be possible to edit your door bell project to not read messages at all though. The basic functionality your described (press button, send picture) does not need to read from Telegram at all, it could just send messages.  I can't really advise on how to do this though.

I'm going to close this issue as it isn't a issue with the library

closed 2020-04-29 by witnessmenow

prontodos, 2020-05-02:
good job thank you for your reply !!