witnessmenow/Universal-Arduino-Telegram-Bot #168: Connection Error

ShrileshKale, 2020-05-02:
@witnessmenow I tried Fingerprint setup using client.setFingerprint() as well as client.setInsecure() ( in case if fingerprint is not setup ) but i always get **connection erro**r.

Setup: Board - 2..6.2  Arduino IDE - 1.8.12 

Previously I tried
2.5.2, 2.4 .2, ,2.6.3 version of ESP8266 core but no success at all.

Could you please help me up?

witnessmenow, 2020-05-02:
Is telegram blocked in your country?

ShrileshKale, 2020-05-02:
No...I live in Germany ...what could be the reason behind it?

witnessmenow, 2020-05-02:
I know someone from Germany was saying before he had some issues with his
ISP when they moved to ipv6 addresses, could you test hot spotting your
phone?

ShrileshKale, 2020-05-02:
@witnessmenow  I tried with hotspot now but still no success.

phschoen, 2020-06-08:
i have the same issue. with an esp8266 its not working. even the basic EchoBot example is not working.
However with the example of an esp32 with the EchoBot is working correctly.
here the logs for the esp8266 with 
  -D CORE_DEBUG_LEVEL=1
  -D DEBUG_ESP_SSL=1

```
BSSL:_connectSSL: start connection
BSSL:Connection *will* fail, no authentication method is setup
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Chain could not be linked to a trust anchor.'
0
BSSL:_connectSSL: start connection
BSSL:Connection *will* fail, no authentication method is setup
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Chain could not be linked to a trust anchor.'
0
pm open,type:2 0
BSSL:_connectSSL: start connection
BSSL:Connection *will* fail, no authentication method is setup
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Chain could not be linked to a trust anchor.'
0
BSSL:_connectSSL: start connection
BSSL:Connection *will* fail, no authentication method is setup
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Chain could not be linked to a trust anchor.'

```

witnessmenow, 2020-06-24:
@phschoen your logs indicate that you don't have a `client.setInsecure();` called in the setup. I still don't know what could be wrong with the original posters problem

phschoen, 2020-06-24:
Ok. Thanks btw. but this is then missing in the ESP8266 examples since there are separate ones for esp32 and ESP8266

witnessmenow, 2020-06-24:
Not as of earlier this week :)

witnessmenow, 2020-06-24:
Well, in the master branch, it's still not released yet

closed 2020-09-30 by witnessmenow