witnessmenow/Universal-Arduino-Telegram-Bot #169: May 2020 - Instructions

jameszah, 2020-05-04:
First, this needs ArduinoJson ver #5, rather than current version #6 so go to Arduino - Tools - Manage Libraries, find ArduinoJson, and change it to the highest release of version 5.  
You can switch it back to 6 after the compile.

Second, there is some security SSL upgrades since this was released, so add the statement 
`  client.setInsecure();`
as the last line of setup().

Or even better, add these lines

```
  const uint8_t fingerprint[20] = { 0xF2, 0xAD, 0x29, 0x9C, 0x34, 0x48, 0xDD, 0x8D, 0xF4, 0xCF, 0x52, 0x32, 0xF6, 0x57, 0x33, 0x68, 0x2E, 0x81, 0xC1, 0x90 };
  client.setFingerprint(fingerprint); 
```

at the end of setup.

Read this post to get the fingerprint: (not sure if these fingerprints change ???)
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/118#issuecomment-585831804

jameszah, 2020-05-11:
Update:

I tried to implement this on esp32, and I see that the setInsecure, and setFingerprint calls to not exist in the esp32 library.  

So the solution is: do nothing -- just leave out the setInsecure and setFingerprint, and it still works.  You just get this message, if you have messages turned on:

`[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!`

fritsjan, 2020-07-20:
How does this compare with v 1.2 of this library?
There we use ArduinoJson v 6 right?

And is there a way to correctly implement certificates? Can this be done with a fingerprint?

jameszah, 2020-07-20:
This May 2020 info was for v1.1 - the current master at the time, which used the ArduinoJson v5.  
I think this is still the version you get from the Arduino Library Manager.

The July 2020 master v1.2 (i think) uses ArduinoJson v6.

Those fingerprint commands above work for 8266, and "do nothing" works for esp32, but I think the proper way is certificates, which I have not pursued.  I saw a few examples searching, but it looked like similar to the fingerprint, but much longer series of numbers.

closed 2020-08-13 by jameszah