witnessmenow/Universal-Arduino-Telegram-Bot #170: Stops working when connecting to a different router

rafaelcbarbosa, 2020-05-11:
I am having a weird problem: everything works fine when my eps8266ex (esp-01s) is connected to my ASUS Router that connects to my modem. However, when I try to connect to an old Access Point, the bot connection stops working and apparently it jam's the entire connection to that Access Point. This access point is connected to the Asus Router. 

I have tried many different versions of the esp8266 core (from 2.4.2 onwards) with no success.

The error I am getting is this:
`20:30:59.796 -> connected with Coxinha-Sala, channel 3
20:30:59.796 -> dhcp client start...
20:30:59.831 -> ip:192.168.20.28,mask:255.255.255.0,gw:192.168.20.1
20:31:00.043 -> BSSL:_connectSSL: start connection
20:31:00.315 -> BSSL:_wait_for_handshake: failed
20:31:00.315 -> BSSL:Couldn't connect. Error = 'Unknown error code.'
20:31:00.315 -> numNewMessages: 0
20:31:01.539 -> BSSL:_connectSSL: start connection
20:31:01.810 -> BSSL:_wait_for_handshake: failed
20:31:01.810 -> BSSL:Couldn't connect. Error = 'Unknown error code.'
20:31:01.810 -> numNewMessages: 0
20:31:03.039 -> BSSL:_connectSSL: start connection
20:31:03.313 -> BSSL:_wait_for_handshake: failed
20:31:03.313 -> BSSL:Couldn't connect. Error = 'Unknown error code.'
20:31:03.313 -> numNewMessages: 0
20:31:04.546 -> BSSL:_connectSSL: start connection
20:31:04.784 -> BSSL:_wait_for_handshake: failed
20:31:04.784 -> BSSL:Couldn't connect. Error = 'Unknown error code.'
20:31:04.784 -> numNewMessages: 0
20:31:05.783 -> state: 5 -> 2 (3c0)
20:31:05.783 -> rm 0
20:31:05.783 -> BSSL:connect: Unable to connect TCP socket
20:31:05.783 -> numNewMessages: 0
20:31:06.793 -> reconnect
20:31:06.793 -> state: 2 -> 0 (0)
20:31:06.793 -> BSSL:connect: Unable to connect TCP socket
20:31:06.793 -> numNewMessages: 0
20:31:06.929 -> scandone
20:31:06.929 -> state: 0 -> 2 (b0)
20:31:06.929 -> state: 2 -> 3 (0)
20:31:06.929 -> state: 3 -> 5 (10)
20:31:06.929 -> add 0
20:31:06.929 -> aid 2
20:31:06.929 -> cnt
20:31:07.794 -> BSSL:connect: Unable to connect TCP socket
20:31:07.794 -> numNewMessages: 0
20:31:07.827 -> `

and my source code (very simple):

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <UniversalTelegramBot.h>

#ifndef STASSID
#define STASSID "Coxinha-Sala"
#define STAPSK  "mercadolivrez"
#define BOTTOKEN "1041453184:AAH4WVWmWLyQ62btyxV5-gWqetUZB_306GY"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;
const char* bot_token = BOTTOKEN;

ESP8266WebServer server(80);

const int ledPin = 2;
WiFiClientSecure client;
UniversalTelegramBot  *bot;
const int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

void handleRoot() {
  digitalWrite(ledPin, LOW);
  server.send(200, "text/plain", "hello from esp8266!");
  delay(300);
  digitalWrite(ledPin, HIGH);
}

void setup(void) {
  pinMode(ledPin, OUTPUT);
  digitalWrite(ledPin, 0);
  Serial.begin(115200);

  startLedSequence();

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");

  client.setInsecure();

  bot = new UniversalTelegramBot(bot_token, client);

  Serial.println(ESP.getFullVersion());
}

void loop(void) {
  server.handleClient();
  MDNS.update();

  if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot->getUpdates(bot->last_message_received + 1) + 0;
    Serial.println("numNewMessages: " + (String)numNewMessages);
    while (numNewMessages) {
      Serial.println("Message received from: " + bot->messages[0].from_id);
//      handleNewMessages(numNewMessages);
      numNewMessages = bot->getUpdates(bot->last_message_received + 1);
    }
    Bot_lasttime = millis();
  }
}

void startLedSequence() {
  Serial.println("Starting...");

  digitalWrite(ledPin, HIGH);  // Turn the LED off by making the voltage HIGH
  Serial.println("ledPin off");

  Serial.println("Starting led sequence completed.");

}

void handleNewMessages(int numNewMessages) {

  Serial.println("Received " + String(numNewMessages) + " new request(s)...");

}

rafaelcbarbosa, 2020-05-17:
[PARTIALLY SOLVED] 

I retried setting up as 160mhz did not work.

But then I tried to force the ESP-01s into I forced it to connect using 802.11B (vs N) and it started working. I tried 802.11G and ended up with the same problem...

In summary, I solved it by force the setup to 802.11B, since I don't need bandwidth, just need stable connection. The performance was not impacted for this case.

The command I used was:

WiFi.setPhyMode(WIFI\_PHY\_MODE\_11B);

Any idea why that could be happening?

witnessmenow, 2020-06-24:
I don't know what could be wrong, and I also doubt this is something that could be fixed by this library.

I would recommend raising an issue with the ESP8266 arduino project, although they would probably want a really basic example that does not involve this library.

closed 2020-06-24 by witnessmenow