witnessmenow/Universal-Arduino-Telegram-Bot #173: Stopped working this morning

thevalo, 2020-06-07:
Today, the UATB doesn't receive messages, nor send them. It worked yesterday though. I have turned the debugging and this is the log:
`12:31:46.908 -> GET Update Messages
12:31:46.908 -> [BOT]Connecting to server
12:31:47.860 -> .... connected to server
12:31:48.193 -> 
12:31:48.193 -> <!DOCTYPE html>
12:31:48.193 -> <html>
12:31:48.193 ->   <head>
12:31:48.228 ->     <meta charset="utf-8">
12:31:48.228 ->     <title>getUpdates</title>
12:31:48.263 ->     <meta name="viewport" content="width=device-width, initial-scale=1.0">
12:31:48.365 ->     <meta property="description" content="">
12:31:48.399 ->     <meta property="og:title" content="getUpdates">
12:31:48.475 ->     <meta property="og:image" content="">
12:31:48.542 ->     <meta property="og:description" content="">
12:31:48.582 ->     <link rel="shortcut icon" href="/favicon.ico?3" type="image/x-icon" />
12:31:48.624 -> 
12:31:48.624 ->     <link href="/css/bootstrap.min.css?3" rel="stylesheet">
12:31:48.694 ->     
12:31:48.694 ->     <link href="/css/telegram.css?198" rel="stylesheet" media="screen">
12:31:48.766 ->     <style>
12:31:48.804 ->     </style>
12:31:48.804 ->   </head>
12:31:48.804 ->   <body class="preload">
12:31:48.856 ->     <div class="dev_page_wrap">
12:31:48.856 ->       <div class="dev_page_head navbar navbar-static-top navbar-tg">
12:31:48.931 ->         <div class="navbar-inner">
12:31:48.974 ->           <div class="container clearfix">
12:31:49.043 ->             <ul class="nav navbar-nav navbar-right hidden-xs"><li class="navbar-twitter"><a href="https://twitter.com/telegram" target="_blank" data-track="Follow/Twitter" onclick="trackDlClick(this, event)"><i class="icon icon-twitter"></i><span> Twitter</span></a></li></ul>
12:31:49.284 ->             <ul class="nav navbar-nav">
12:31:49.361 ->               <li><a href="//telegram.org/">Home</a></li>
12:31:49.395 -> <li class="hidden-xs"><a href="//telegram.org/faq">FAQ</a></li>
12:31:49.465 -> <li class="hidden-xs"><a href="//telegram.org/apps">Apps</
12:31:49.535 -> 
12:31:49.535 -> incoming message length1300
12:31:49.569 -> Creating DynamicJsonBuffer
12:31:49.603 -> Failed to parse update, the message could be too big for the buffer
12:31:49.672 -> Closing client`

Seems like something might have changed in Telegram servers, as it's receiving another response.
According to some other users in Telegram's group, it stopped working for them too, so I'd consider it's a general problem.

DrXoo, 2020-06-07:
Same here, it stopped working at 00:00 of today

denzen84, 2020-06-07:
+1, esp32, doesn't work via Tor WiFi access point

witnessmenow, 2020-06-07:
Pushed a fix for this to V1.2 of the library, this branch also uses ArduinoJsonV6 so you will need to update that too.

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.2.0

If you want to stay on the same version, It's possible that you can just update the sendGetToTelegram method in the .cpp with this: https://gist.github.com/witnessmenow/5ddb52d810122d068b6d81b1ac625ce1

But it's untested

gerasdf, 2020-06-07:
See https://github.com/gerasdf/Universal-Arduino-Telegram-Bot/issues/11
It's fixed there

potonzio, 2020-06-07:
I have just tested the fix in 1.1.0 library, it works.

thevalo, 2020-06-07:
> Pushed a fix for this to V1.2 of the library, this branch also uses ArduinoJsonV6 so you will need to update that too.
>
> https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/V1.2.0
>
> If [...]

The V1.2 has fixed the issue, however what I've noticed is that whether debug is set to false or not declared at all, (`bot._debug = false;`), it still prints debug messages in the serial monitor. Any idea why?

witnessmenow, 2020-06-07:
Yeah sorry about that, I'm moving away from the boolean debug flag and making it compiler level #define whether to include it or not. This means if you don't want it in your project, the serial prints will get removed at compile time.

I left it in by mistake when uploading the new version. I'll update the V1.2 branch removing it, but if you want to save yourself the hassle of re-downloading, just remove the following line from the top of libraries UniversalTelegramBot.h file

```
#define TELEGRAM_DEBUG 1
```

closed 2020-06-07 by thevalo

fmarzocca, 2020-06-09:
Yesterday my BOT has started working again, without applying the new version 1.2.
Is it anyway worth to upgrade to 1.2?