witnessmenow/Universal-Arduino-Telegram-Bot #177: Bot can't receive messages

sandroxGH, 2020-06-07:
For the handler i receive this message
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for api.telegram.org
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():273]: (-80) UNKNOWN ERROR CODE (0050)

phschoen, 2020-06-08:
i have the same issue. with an esp8266 its not working. even the basic EchoBot example is not working.
However with the example of an esp32 with the EchoBot is working correctly.

sandroxGH, 2020-06-08:
I've solved enabling the Bot to chat in the group. Im my case the bot is a
remote control for some stuff and is working in a chat group of user.
After enabling it in the bot father  and wait some hours  it start work
like usual.
Regards

sandroxGH, 2020-06-08:
I've solved enabling the Bot to chat in the group. Im my case the bot is a remote control for some stuff and is working in a chat group of user.

closed 2020-06-08 by sandroxGH

Europaco, 2023-01-06:
Hy‼️

I have that same problem:

[WiFiGeneric.cpp:654] hostByName(): DNS Failed for api.telegram.org

[WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1

What do you mean by enabling it in the botFather?
What command do you use in the father bot to reactivate?