witnessmenow/Universal-Arduino-Telegram-Bot #181: send .txt file

droidprova, 2020-06-18:
Hello and thank you for the precious work you do. I noticed that you are working on updating this fantastic library. I have a request for you: is it possible to add a function that allows you to send a .txt file stored on an SD card? Exactly how it happens for photos?
Thanks so much

witnessmenow, 2020-06-24:
Will take a look if I have time.

solcer, 2020-09-13:
> Hello and thank you for the precious work you do. I noticed that you are working on updating this fantastic library. I have a request for you: is it possible to add a function that allows you to send a .txt file stored on an SD card? Exactly how it happens for photos?
> Thanks so much

You can also use telegramOta example for send files into spiffs.

droidprova, 2020-09-14:
Hi, could you please see an example. Thank you

solcer, 2020-09-14:
> Hi, could you please see an example. Thank you
Hi,

You can use telegramOTA.ino code for esp32. it is under the examples/Untested inc Upgrade/ESP32/telegram OTA/  folder.

if you send files with "write spiffs" caption, the code will write your file into to spiffs.
If  you send spiffs.bin file with "update spiffs" caption: it writes spiffs image to spiffs section.
If you send firmware.bin file with "update firmware" caption:  it updates firmware.

Besst

PoriadnyyOleg, 2021-01-23:
Hi! To send txt file by this library need make some changes: 
Make a copy of the sendPhotoByBinary and replace the string 
`String response = sendMultipartFormDataToTelegram("sendPhoto", "photo", "img.jpg",` 
to 
`String response = sendMultipartFormDataToTelegram("sendDocument", "document","file.txt",` 
I tried it, it works and sends me the "file.txt".
Also i change "file.txt" to new input var String& fileName.

PoriadnyyOleg, 2021-01-23:
Hi! To send a .txt file stored on an SD card by this library need some changes: 
Make a copy of sendPhotoByBinary and change string:
`String response = sendMultipartFormDataToTelegram("sendPhoto", "photo", "img.jpg",`
to 
`String response = sendMultipartFormDataToTelegram("sendDocument", "document", "file.txt",`
I tried it, it works and sends me a "file.txt".
Also i change "file.txt" to new input variable String& fileName.

todosko, 2021-08-01:
I want to check the telegramOTA.ino example, but could not figure out what the message format should be in the Telegram line. And what is "file caption" ?! (my english is lame) Is that the name of the file?
For example: I want to update the firmware with the BLINK.bin file on the C: drive in the BLINK directory. How to write a message correctly?
The / dir and / format commands work well.

Emb-Coder, 2022-07-05:
> Hello and thank you for the precious work you do. I noticed that you are working on updating this fantastic library. I have a request for you: is it possible to add a function that allows you to send a .txt file stored on an SD card? Exactly how it happens for photos? Thanks so much

Hi did you find any working example ?

akil1947, 2023-06-02:
> Hi! To send a .txt file stored on an SD card by this library need some changes: Make a copy of sendPhotoByBinary and change string: `String response = sendMultipartFormDataToTelegram("sendPhoto", [...]
can you please elaborate on this. If possible please give me the full code

PoriadnyyOleg, 2023-06-02:
add in .h :
String sendDocumentByBinary(
    const String& chat_id, const String& contentType, int fileSize,
    MoreDataAvailable moreDataAvailableCallback,
    GetNextByte getNextByteCallback, GetNextBuffer getNextBufferCallback, GetNextBufferLen getNextBufferLenCallback, const String& fileName);

add in .cpp:
String UniversalTelegramBot::sendDocumentByBinary(
    const String& chat_id, const String& contentType, int fileSize,
    MoreDataAvailable moreDataAvailableCallback,
    GetNextByte getNextByteCallback, GetNextBuffer getNextBufferCallback, GetNextBufferLen getNextBufferLenCallback, const String& fileName) {

  #ifdef TELEGRAM_DEBUG  
    Serial.println(F("sendDocument: SEND Document"));
  #endif

  String response = sendMultipartFormDataToTelegram("sendDocument", "document", fileName,
    contentType, chat_id, fileSize,
    moreDataAvailableCallback, getNextByteCallback, getNextBufferCallback, getNextBufferLenCallback);

  #ifdef TELEGRAM_DEBUG  
    Serial.println(response);
  #endif

  return response;
}

akil1947, 2023-06-04:
@PoriadnyyOleg thank you so much for your help!