witnessmenow/Universal-Arduino-Telegram-Bot #183: Can this library actively push notification to user?

nkebx2, 2020-06-24:
Good day, thanks for sharing this library. I tried to measure the temperature of my server room with some temperature sensors and I wanted the bot to actively inform me if the server room exceeded certain temperature. I wonder this could be done in this library, thanks!

witnessmenow, 2020-06-24:
yup, use this example as your base:
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP32/SendMessageFromEvent/SendMessageFromEvent.ino

Just move the sendMessage call from the setup to be inside some logic for checking temp

```
if(temp > 25){
    bot.sendMessage(CHAT_ID, "Too hot!", "");
    delay(1000 * 60); //Do something to stop it repeating the message over and over again. This is a 60 second delay, but you probably should do something smarter than this
}
```

closed 2020-06-24 by witnessmenow

nkebx2, 2020-06-25:
Thanks Brian, please keep up the good work!