witnessmenow/Universal-Arduino-Telegram-Bot #184: (info) Send Caption with Picture

jameszah, 2020-06-24:
Here is a bit of code to send a caption with a picture.  There are a lot of branches, and a new master, and I'm not sure how these forks and pulls work, so I thought I would just send it as text.

First, you copy the function sendMultipartFormDataToTelegram, and rename it sendMultipartFormDataToTelegramWithCaption, then add a parameter "String caption",  add it to the include file, and then add the following to the big start_request string:

```
    start_request = start_request + "\r\n";
    start_request = start_request + chat_id + "\r\n";

    start_request = start_request + "--" + boundry + "\r\n";                              //jz caption stuff
    start_request = start_request + "content-disposition: form-data; name=\"caption\"" + "\r\n";
    start_request = start_request + "\r\n";
    start_request = start_request + caption + "\r\n";                                          //jz end

    start_request = start_request + "--" + boundry + "\r\n";
    start_request = start_request + "content-disposition: form-data; name=\"" + binaryProperyName + "\"; filename=\"" + fileName + "\"" + "\r\n";

```

Those four lines in the middle above are new to send the caption string.

Then in your code, you call it with 

```
  String sent = bot.sendMultipartFormDataToTelegramWithCaption("sendPhoto", "photo", "img.jpg",
    "image/jpeg", the_caption, BOTme, fb_length,
    isMoreDataAvailable, getNextByte, nullptr, nullptr);

```
where BOTme is your telegram id, and the_caption is a big string of whatever you want -- here I am sending various parameters about an ESP32-CAM -- just an example of sending lots of data as text with the picture.

```
  char the_caption[500];

  int tot = SD_MMC.totalBytes() / (1024 * 1024);
  int use = SD_MMC.usedBytes() / (1024 * 1024);
  long rssi = WiFi.RSSI();

  const char msg[] PROGMEM = R"rawliteral(
  ESP32-CAM Video Recorder
   New video started:
   Used %d MB / %d MB, Rssi %d
  )rawliteral";

  sprintf(the_caption, msg, use, tot, rssi );
```

Also, as discussed in this issue https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/68, I have added the code 

`Serial.print("*")  ; delay(100);`

after every client->println and client->write in this function.  I assume this is something to do with telegram controlling bursts of messages, but it works with this, and does not work without it on my ESP32-CAM application sending a photo from memory.   This is not related to the caption business, but another change from Branch V1.2 of a couple weeks ago.  I tried the new master, it didn't work with my existing code -- and haven't pursued it yet.

You send reply with questions within the issue.

witnessmenow, 2020-06-24:
Thank for the info, will take a look at fixing the issue described in #68 for sure, and I will see about including this.

jameszah, 2020-08-09:
This was a mod of v1.1.
Look here for a demo, that uses v1.2 - the current master.

https://github.com/jameszah/ESP32-Cam-Telegram

closed 2020-08-14 by jameszah