witnessmenow/Universal-Arduino-Telegram-Bot #186: (bugfix) Sending photos with new Master July 2020

jameszah, 2020-07-09:
In the function sendMultipartFormDataToTelegram, line 239 of UniversalTelegramBot.cpp, change

`client->print(F("Host: " TELEGRAM_HOST));`

... to ...

`client->println(F("Host: " TELEGRAM_HOST));`

Also, when using

```
  String sendPhotoByBinary(const String& chat_id, const String& contentType, int fileSize,
                           MoreDataAvailable moreDataAvailableCallback,
                           GetNextByte getNextByteCallback, 
                           GetNextBuffer getNextBufferCallback, 
                           GetNextBufferLen getNextBufferLenCallback);

```
the final two parameters, which send the entire photo as one "write" can only be used for smaller photos -- it works in my debugging with a 4 kb photo -- but if you are sending larger photos like a UXGA 60kb+ you have to use the previous two paramters  MoreDataAvailable and GetNextByte, which send the photo as a series of 512 byte writes to the wifi.  In another post I talked about putting a delay between each write, but today I reduced my delay to zero and everything works fine, although I still did a debug write statement to serial after each wifi write ... giving some delay.

I'll post my ESP32-Cam demo sketch with UniversalTelegramBot.cpp with my new line 239 if anyone is interested.

witnessmenow, 2020-07-09:
I'd be interested in seeing the demo sketch, thanks for the suggestion for
the 239 thing

jameszah, 2020-07-10:
https://github.com/jameszah/ESP32-Cam-Telegram

fritsjan, 2020-07-29:
Added this fix in pull-request #190 , thanks @jameszah

closed 2020-08-13 by jameszah