witnessmenow/Universal-Arduino-Telegram-Bot #187: More than one person allowed to send commands.

Seruch, 2020-07-11:
Hi, 
Please help me with code allowing me and my girlfriend to send commands to out bot that is able to feed cats when we are away. For now i have something like this:
    if ( chat_id != "1234567890 ) {
      bot.sendMessage(chat_id, "You are now allowed to feed our cats", "");
    }
    else if ( chat_id == "1234567890"){
    if (text == "/feed") {

And rest of code that let me feed cats. We need to be able to do this both. Adding this like this:

    if ( chat_id != "1234567890 || 1234567890" ) 

Does not work, can someone please show me way to add another ID?

gerasdf, 2020-07-11:
```C
if ( chat_id != "1234567890 || 1234567890" ) 
```
should be written as

```C
if ( (chat_id != "1234567890") && (chat_id != "1234567890" ))
```
I suggest that you put that logic in a function and then use it as in:

```C
#define MY_ID    "12345678"
#define GF_ID    "87654321"

bool is_from_friend(String chat_id) {
   return (
     (chat_id == MY_ID) ||
     (chat_id == GF_ID)
   )
}
...

if (is_from_friend(chat_id) {
   if (text == "/feed") {
   ...
   }
} else {
 bot.sendMessage(chat_id, "You are now allowed to feed our cats (" + chat_id + ")", "");
}
```

Seruch, 2020-07-14:
Thanks, im not programmer at all, so most of work with code i try to copy and paste other people solutions. This:
if ( (chat_id != "1234567890") && (chat_id != "1234567890" ))
Makes bot go crazy - when i send command to him, my gf gets response. If she send commands bot does not respond at all and do nothing. I guess that its messed up.  Sadly other solution that you gave is beyond my skill. From what i understand i should put this:
#define MY_ID    "12345678"
#define GF_ID    "87654321"

On top of code with rest of  "#define "1232134" section. 

But next no matter where i put:
bool is_from_friend(String chat_id) {
   return (
     (chat_id == MY_ID) ||
     (chat_id == GF_ID)
   )
}
I get some error "there should be ; before {" and it just wont let me upload it into NodeMCU. I read "bool" section on arduino but to no avail - every example is simple as "bool xxx ; yyy". 
Anyway thanks for help, gonna try to solve this bit more, but if you could send some tips it would be greatful. Maybe all of my code that control bot woudl help: 

void handleNewMessages(int numNewMessages) {
  Serial.println("New Messages");
//  Serial.println(String(numNewMessages));

  for (int i=0; i<numNewMessages; i++) {
    String chat_id = String(bot.messages[i].chat_id);
    String text = bot.messages[i].text;

    String from_name = bot.messages[i].from_name;
    if (from_name == "") from_name = "MyName";
    if ( chat_id != "12345678") {
      bot.sendMessage(chat_id, "You have no power here", "");
    }
    else if ( chat_id == "12345678"){
    if (text == "/ip") {
        String CatFeeder = WiFi.localIP().toString();
        bot.sendMessage(chat_id, "IP CatFeeder: " + (CatFeeder), "");
      }
    
    if (text == "/feed") {
          feedCats();
          bot.sendMessage(chat_id, "Done", "");
      }
      
    if (text == "/start") {
      String welcome = "Hello, " + from_name + ".\n\n";
      welcome += "feeder bot\n\n";
      welcome += "/ip :  IP\n";
      welcome += "/feed : feed \n";
      bot.sendMessage(chat_id, welcome, "Markdown");
    }
   }
  }
}

witnessmenow, 2020-09-30:
Not really a library issue.

closed 2020-09-30 by witnessmenow