witnessmenow/Universal-Arduino-Telegram-Bot #189: [Feature Request] Async Universal Telegram Bot Library

gnalbandian, 2020-07-24:
Hi @witnessmenow. Nice library you have build up here.

I'm using it in a huge and very complex project where things happen fast and I have to take measurements and other stuff as fast as possible.

I was wondering if there is any step forward in making the library work in asynchronous mode rather than the way in which it works now.

Sometimes, waiting to the connection to be established, or waiting for a response from telegram servers, just takes to much and wastes precious time that could be used in other tasks.

Let me know what you think.

gerasdf, 2020-07-24:
Uhm. I've been looking a bit at this sort of speed too, but haven't had time to make much testing. My first idea to try though, is to keep the client connection open, instead of closing it and re-opening each time, I do know that part of the delays is due to reopening it. I know it may consume some extra RAM, but if you application allows it, it may be a good idea to try.

let me know please!

gnalbandian, 2020-07-25:
For starters I believe an async TCP library should be used. Maybe [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP).
Ofcourse this would meant some code refactoring to use this library.

RomeHein, 2020-07-29:
Same needs here for [ESPecial](https://github.com/RomeHein/ESPecial).
Atm I take measurement of all I/O on core 0 of the ESP32 and do the rest (telegram bot) on core 1. It's not ideal but it's fast enough.

gnalbandian, 2020-08-06:
@witnessmenow any comments on this?

witnessmenow, 2020-08-24:
AsyncTCP doesn't look suitable anyways, it doesn't seem to be a HTTP client.

I'm also not a big fan of making the library be dependent on other external libraries where it can be avoided.

jamesmyatt, 2020-08-27:
There's https://github.com/boblemaire/asyncHTTPrequest for async HTTP requests on ESP, but the objective here is for a universal library, i.e. all major platforms. So I think that this can only be achieved in a new library that is ESP only.

gnalbandian, 2020-08-27:
Hi @jamesmyatt. Thanks for sharing that library. I'll look into it.
Regarding the universality of this library, I belive it's blocking behavior is as universal as it can be.
All platform will suffer from the blocking as that is the way the library is intended to work, regardless of the platform.
There are specific platform where the impact of this can be unnoted, like ESP32 where main loop runs in one core, and telegarm loop runs on the second, but its blocking behavior still remains.

jamesmyatt, 2020-08-27:
Actually, asyncHTTPrequest only supports HTTP and not HTTPS, which Telegram requires. HTTPS on ESP8266 is generally not great.

witnessmenow, 2020-09-30:
I'm going to close this for now, I dont see any easy solution for it at the moment

closed 2020-09-30 by witnessmenow