witnessmenow/Universal-Arduino-Telegram-Bot #192: take and send photo as 2 separate actions

mugginsjm, 2020-08-15:
Hi before the library update I was able to do the 2 actions..
Can anyone advise me how to do it now. I have a situation where one trigger takes a photo and then dependent on further 2 triggrers I send to ftp or Telegram. 
Thanks 
John

void takePhoto(){
 fb_length = NULL;
fb_buffer = NULL;
camera_fb_t * fb = NULL;
fb = esp_camera_fb_get();
fb_length = fb->len;
fb_buffer = fb->buf;
esp_camera_fb_return(fb);
}

void sendPhoto(String chat_id,){  
  debugPL("starting telegram upload");
currentByte = 0;
bot->sendPhotoByBinary(chat_id, "image/jpeg", fb_length, isMoreDataAvailable, photoNextByte, nullptr, nullptr);

}

jameszah, 2020-08-16:
Two things:

1. You are "returning" or deleting the picture as soon as you take it.  It might still be sitting in memory, or it might not.  So make make sure you do the "sendPhoto..." before you call the fb_return.  You can keep the "fb" pointer in a global variable and do it later, and then call the fb_return when you are done with it.

2.  There is a bug in the sendPhotoByBinary described here: https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/186

I don't think the fix has been included in the master library yet.

... and there is another issue with sendPhotoByBinary that happens (sometimes) when you are sending very large photos described here: https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/68

You can look here https://github.com/jameszah/ESP32-Cam-Telegram for a fully working demo and clip out that version of the library that works for photos.

mugginsjm, 2020-08-17:
Great.. thanks for the pointers.  I'll have a look.

jameszah, 2020-08-17:
Something like this -- where you set fb in takePhoto() and then you use it in sendPhoto().  

If you do not send the photo, you will have to call esp_camera_fb_return(fb); at some point to get rid of the previous photo.

```
uint8_t* fb_buffer = NULL;
size_t fb_length = NULL;
camera_fb_t * fb = NULL;

void takePhoto() {
  fb = esp_camera_fb_get();
}

void sendPhoto(String chat_id,) {
  debugPL("starting telegram upload");

  fb_length = fb->len;
  fb_buffer = fb->buf;

  currentByte = 0;
  bot->sendPhotoByBinary(chat_id, "image/jpeg", fb_length, isMoreDataAvailable, photoNextByte, nullptr, nullptr);

  esp_camera_fb_return(fb);
}
```

witnessmenow, 2020-09-30:
I think @jameszah has pointed in the right direction with this

closed 2020-09-30 by witnessmenow