witnessmenow/Universal-Arduino-Telegram-Bot #197: Make BOT_TOKEN changeble

jowiemann, 2020-08-27:
Hello,

with a little change it would be possible to chnage bot token via runtime:

**UniversalTelegramBot.h**
UniversalTelegramBot(Client &client);
void setTelegramBotToken(String token);

**UniversalTelegramBot.cpp**
UniversalTelegramBot::UniversalTelegramBot(Client &client) {
  _token = "";
  this->client = &client;
}

void UniversalTelegramBot::setTelegramBotToken(String token) {
  _token = token;
}

Thank you

jamesmyatt, 2020-08-27:
There's an updateToken method. Does that work?

witnessmenow, 2020-08-27:
as mentioned by @jamesmyatt , updateToken is exactly what you are looking for

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.cpp#L46

closed 2020-08-27 by witnessmenow