witnessmenow/Universal-Arduino-Telegram-Bot #198: Test all examples since the upgrade of the library

witnessmenow, 2020-08-28:
The current version of the library is quite different than the previous version so the examples need to be tested to check they still work

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/master/examples/UntestedSinceUpgrade

If anybody would like to help do this I would really appreciate it. Please use the [current examples](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/master/examples) as a basis (there are some notes on the esp8266 version that I think are important)

droidprova, 2020-09-08:
Hi, I only have esp8266 boards, tell me what I can concretely do to help you in development, and if I am able I will gladly help you.
Hello

Bolukan, 2020-10-03:
Contributed #201

Anton-V-K, 2020-11-08:
The link in the initial comments points to a non-existent page - 404 is returned by github.
In the current examples folder only _telegramOTA_ is marked as _untested_.