witnessmenow/Universal-Arduino-Telegram-Bot #208: Severe Heap Fragmentation

Anton-V-K, 2020-10-22:
I tried to use the library in my project, and noticed that `getUpdates` causes (indirectly?) critical memory fragmentation, so my sketch fails to function properly (there are other parts like Web-server, HTTPS-requests, etc. which rely on dynamic memory).

And the issue is actually reproducible with EchoBot example - you just need to add memory checks around `getUpdates` (refer to [guidelines](https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html)).
The final `loop` will look like this:
```
void loop() {
  if (millis() > lastTimeBotRan + botRequestDelay)  {
    
    Serial.printf("FreeHeap = %d, HeapFragmentation = %d, MaxFreeBlockSize = %d\n"
                , ESP.getFreeHeap(), ESP.getHeapFragmentation(), ESP.getMaxFreeBlockSize());
                
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    
    Serial.printf("FreeHeap = %d, HeapFragmentation = %d, MaxFreeBlockSize = %d\n"
                , ESP.getFreeHeap(), ESP.getHeapFragmentation(), ESP.getMaxFreeBlockSize());

    while (numNewMessages) {
      Serial.println("got response");
      for (int i = 0; i < numNewMessages; i++) {
        bot.sendMessage(bot.messages[i].chat_id, bot.messages[i].text, "");
      }
      // numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }

    lastTimeBotRan = millis();
  }
}
```

This is how execution goes:
```
02:48:32.309 -> WiFi connected
02:48:32.309 -> IP address: 192.168.1.64
02:48:32.309 -> FreeHeap = 44216, HeapFragmentation = 5, MaxFreeBlockSize = 42376
02:48:33.431 -> FreeHeap = 43808, HeapFragmentation = 4, MaxFreeBlockSize = 42376
02:48:34.449 -> FreeHeap = 43944, HeapFragmentation = 4, MaxFreeBlockSize = 42376
02:48:36.965 -> FreeHeap = 43616, HeapFragmentation = 3, MaxFreeBlockSize = 42376
02:48:37.951 -> FreeHeap = 43944, HeapFragmentation = 4, MaxFreeBlockSize = 42376
02:48:39.004 -> FreeHeap = 43616, HeapFragmentation = 3, MaxFreeBlockSize = 42376
02:48:40.024 -> FreeHeap = 43752, HeapFragmentation = 4, MaxFreeBlockSize = 42376
02:48:42.607 -> FreeHeap = 42920, HeapFragmentation = 31, MaxFreeBlockSize = 21736
02:48:43.590 -> FreeHeap = 43248, HeapFragmentation = 32, MaxFreeBlockSize = 21736
```
So in about 10 seconds more a lot of RAM becomes unavailable because of fragmentation.
Such memory state is kept forever.

Details:
- Arduino IDE 1.8.13 + ESP8266 Core 2.7.4
- UniversalTelegramBot 1.2.0
- ArduinoJson 6.16.1

Anton-V-K, 2020-10-24:
Verbose execution output with `TELEGRAM_DEBUG` enabled:
```
16:14:53.143 -> WiFi connected
16:14:53.143 -> IP address: 192.168.1.64
16:14:53.143 -> FreeHeap = 44104, HeapFragmentation = 2, MaxFreeBlockSize = 43360
16:14:53.143 -> GET Update Messages
16:14:53.177 -> [BOT]Connecting to server
16:14:53.721 -> .... connected to server
16:14:54.265 -> 
16:14:54.265 -> {"ok":true,"result":[]}
16:14:54.265 -> 
16:14:54.265 -> incoming message length 23
16:14:54.303 -> Creating DynamicJsonBuffer
16:14:54.303 -> GetUpdates parsed jsonDoc: {"ok":true,"result":[]}
16:14:54.303 -> GetUpdates parsed jsonObj: {"ok":true,"result":[]}
16:14:54.303 -> no new messages
16:14:54.303 -> Closing client
16:14:54.303 -> FreeHeap = 43696, HeapFragmentation = 1, MaxFreeBlockSize = 43360
16:14:55.288 -> FreeHeap = 43832, HeapFragmentation = 2, MaxFreeBlockSize = 43360
16:14:55.288 -> GET Update Messages
16:14:55.288 -> [BOT]Connecting to server
16:14:55.843 -> .... connected to server
16:14:57.814 -> Received empty string in response!
16:14:57.814 -> Closing client
16:14:57.814 -> FreeHeap = 43504, HeapFragmentation = 3, MaxFreeBlockSize = 42200
16:14:58.800 -> FreeHeap = 43832, HeapFragmentation = 2, MaxFreeBlockSize = 43360
16:14:58.834 -> GET Update Messages
16:14:58.834 -> [BOT]Connecting to server
16:14:59.343 -> .... connected to server
16:14:59.819 -> 
16:14:59.819 -> {"ok":true,"result":[]}
16:14:59.819 -> 
16:14:59.819 -> incoming message length 23
16:14:59.819 -> Creating DynamicJsonBuffer
16:14:59.819 -> GetUpdates parsed jsonDoc: {"ok":true,"result":[]}
16:14:59.819 -> GetUpdates parsed jsonObj: {"ok":true,"result":[]}
16:14:59.819 -> no new messages
16:14:59.819 -> Closing client
16:14:59.819 -> FreeHeap = 43504, HeapFragmentation = 3, MaxFreeBlockSize = 42200
16:15:00.839 -> FreeHeap = 43640, HeapFragmentation = 1, MaxFreeBlockSize = 43360
16:15:00.839 -> GET Update Messages
16:15:00.839 -> [BOT]Connecting to server
16:15:01.349 -> .... connected to server
16:15:03.285 -> Received empty string in response!
16:15:03.285 -> Closing client
16:15:03.285 -> FreeHeap = 42808, HeapFragmentation = 33, MaxFreeBlockSize = 20768
16:15:04.306 -> FreeHeap = 43136, HeapFragmentation = 30, MaxFreeBlockSize = 23168
16:15:04.306 -> GET Update Messages
16:15:04.306 -> [BOT]Connecting to server
16:15:04.884 -> .... connected to server
16:15:05.394 -> 
16:15:05.394 -> {"ok":true,"result":[]}
16:15:05.394 -> 
16:15:05.394 -> incoming message length 23
16:15:05.394 -> Creating DynamicJsonBuffer
16:15:05.394 -> GetUpdates parsed jsonDoc: {"ok":true,"result":[]}
16:15:05.394 -> GetUpdates parsed jsonObj: {"ok":true,"result":[]}
16:15:05.394 -> no new messages
16:15:05.394 -> Closing client
16:15:05.394 -> FreeHeap = 42808, HeapFragmentation = 33, MaxFreeBlockSize = 23168
```

Anton-V-K, 2020-10-24:
After a lot of investigations I tend to think the issue is implicitly caused by the specific implementation of `WiFiClientSecure` (for ESP8266) - `UniversalTelegramBot::closeClient()` calls `WiFiClientSecure::stop()` which may keep some stuff (SSL-related?) in memory (for future reuse?).
Luckily I've discovered **a kind of workaround**: if you increase the polling interval (3000 ms works fine with EchoBot), the memory doesn't become too fragmented.
Here is the execution log:
```
18:27:15.867 -> .......
18:27:20.176 -> WiFi connected
18:27:20.176 -> IP address: 192.168.1.64
18:27:20.176 -> INIT: FreeHeap = 44200, HeapFragmentation = 2, MaxFreeBlockSize = 43456
18:27:20.176 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:20.176 -> FreeHeap = 44200, HeapFragmentation = 2, MaxFreeBlockSize = 43456
18:27:21.229 -> FreeHeap = 43792, HeapFragmentation = 1, MaxFreeBlockSize = 43456
18:27:24.247 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:24.247 -> FreeHeap = 43928, HeapFragmentation = 2, MaxFreeBlockSize = 43456
18:27:25.333 -> FreeHeap = 43600, HeapFragmentation = 3, MaxFreeBlockSize = 42384
18:27:28.356 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:28.356 -> FreeHeap = 43736, HeapFragmentation = 1, MaxFreeBlockSize = 43456
18:27:29.444 -> FreeHeap = 43408, HeapFragmentation = 6, MaxFreeBlockSize = 41168
18:27:32.433 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:32.433 -> FreeHeap = 43040, HeapFragmentation = 5, MaxFreeBlockSize = 41168
18:27:33.588 -> FreeHeap = 42712, HeapFragmentation = 4, MaxFreeBlockSize = 41168
18:27:36.577 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:36.577 -> FreeHeap = 42848, HeapFragmentation = 4, MaxFreeBlockSize = 41168
18:27:37.665 -> FreeHeap = 42520, HeapFragmentation = 4, MaxFreeBlockSize = 41168
18:27:40.689 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:40.689 -> FreeHeap = 42656, HeapFragmentation = 4, MaxFreeBlockSize = 41168
18:27:41.843 -> FreeHeap = 42328, HeapFragmentation = 6, MaxFreeBlockSize = 39928
18:27:44.830 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:44.830 -> FreeHeap = 42464, HeapFragmentation = 6, MaxFreeBlockSize = 39928
18:27:45.917 -> FreeHeap = 42136, HeapFragmentation = 6, MaxFreeBlockSize = 39736
18:27:48.940 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:48.940 -> FreeHeap = 42464, HeapFragmentation = 7, MaxFreeBlockSize = 39736
18:27:49.957 -> FreeHeap = 42136, HeapFragmentation = 6, MaxFreeBlockSize = 39736
```
Approx. only 2KB is "inaccessible" with such approach.
Any incoming message causes _temporary_ memory fragmentation, which is cleaned at the next cycle:
```
18:27:52.949 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:52.949 -> FreeHeap = 42464, HeapFragmentation = 7, MaxFreeBlockSize = 39736
18:27:54.000 -> FreeHeap = 21456, HeapFragmentation = 12, MaxFreeBlockSize = 18992
18:27:54.000 -> got response
18:27:57.769 -> ~~~~~~~ +3000 ms ~~~~~~~
18:27:57.769 -> FreeHeap = 42464, HeapFragmentation = 7, MaxFreeBlockSize = 39736
18:27:58.821 -> FreeHeap = 42136, HeapFragmentation = 6, MaxFreeBlockSize = 39736
```
Choosing longer polling interval makes bot less responsive, but doesn't improve memory fragmentation.

The proper fix at the library level would probably require re-creating the client after each poll cycle, but this doesn't look efficient.

droidprova, 2021-02-07:
hello, I also encountered your same problem, which persists even using a 3000 ms longpoll. Are there better solutions?
Thank you

fmarzocca, 2021-02-07:
> hello, I also encountered your same problem, which persists even using a 3000 ms longpoll. Are there better solutions?
> Thank you

I always had this problem, and I have fixed with RomanLut patch here: https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/143

The solution would be a non-blocking Long Poll, as I asked in that ticket.

droidprova, 2021-02-07:
Thanks fmarzocca (italian? Me too) but the roman patch doesn't work with the latest release 1.3.0 right? Sorry I can't use v1.3.0.

fmarzocca, 2021-02-07:
I am using previous version, not 1.3.0. Otherwise you need to edit the patch.

fmarzocca, 2021-02-07:
The patch completely replace the library, so do not care the version. Of course, the patch is based on previous version.

droidprova, 2021-02-07:
so there is no compatibility with arduino json 6. right?

fmarzocca, 2021-02-07:
Right