witnessmenow/Universal-Arduino-Telegram-Bot #209: ESP32 example sketches do not connect to WiFi when using WiFiClientSecure.h

gon0, 2020-10-23:
Dear community,
i have built many project using this library and I think this is great work! Thank you for this library.

After some time, I started a new project, using the new examples. I have a WPA2 Wifi, which works fine with existing projects and telegram bots.

When I change the actual examples to comment out the "secure" parts, the ESP32 connects to Wifi:

```
#include <WiFi.h>
#include <WiFiClient.h>  // if this is active, the ESP32 connects to Wifi
//#include <WiFiClientSecure.h> // if this is active, the ESP32 does not connect to Wifi
#include <UniversalTelegramBot.h>

// Wifi network station credentials
#define WIFI_SSID "xxx"
#define WIFI_PASSWORD "yyy
// Telegram BOT Token (Get from Botfather)
#define BOT_TOKEN "yyy:xxxxxx"

const unsigned long BOT_MTBS = 1000; // mean time between scan messages
const int ledPin = 22;

WiFiClient secured_client;
//WiFiClientSecure secured_client;
UniversalTelegramBot bot(BOT_TOKEN, secured_client);

```

By "ESP32 not connecting to Wifi", i mean it gets stuck at the marked part, when #include <WiFiClientSecure.h>  is used:

```
void setup()
{
  Serial.begin(115200);
  Serial.println();

  // attempt to connect to Wifi network:
  Serial.print("Connecting to Wifi SSID ");
  Serial.print(WIFI_SSID);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  //secured_client.setCACert(TELEGRAM_CERTIFICATE_ROOT); // Add root certificate for api.telegram.org
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");    // here, the ESP32 is stuck forever
    delay(500);
  }
  
  Serial.print("\nWiFi connected. IP address: ");
  Serial.println(WiFi.localIP());
```

Does anyone else experience this problem?

Anton-V-K, 2020-10-24:
This is rathere strange behavior...
Which version of ESP32 SDK do you use?

If you remove references to UniversalTelegramBot, does the issue still happen?
I mean these lines:
```
#include <UniversalTelegramBot.h>
...
UniversalTelegramBot bot(BOT_TOKEN, secured_client);
```
And which version of this library do you use?

Technically `WiFiClientSecure` is just derived from `WiFiClient`, and `WiFi` is always of  type `WiFiClass`.
The constructor `WiFiClientSecure` is quite heavy (if compared to `WiFiClient`) since it initializes SSL part:
```
WiFiClientSecure::WiFiClientSecure()
{
    _connected = false;

    sslclient = new sslclient_context;
    ssl_init(sslclient);
    sslclient->socket = -1;
    sslclient->handshake_timeout = 120000;
    _CA_cert = NULL;
    _cert = NULL;
    _private_key = NULL;
    _pskIdent = NULL;
    _psKey = NULL;
    next = NULL;
}
```
I guess, the issue is related to ESP32 SDK.

gon0, 2020-10-24:
Dear Anton,

thank you for your reply. 

via 

```
  Serial.println("Using ESP object:");
  Serial.println(ESP.getSdkVersion());
 
  Serial.println("Using lower level function:");
  Serial.println(esp_get_idf_version());
```
I am getting this result for the ESP32 SDK:

```
Using ESP object:
v3.2.3-14-gd3e562907
Using lower level function:
v3.2.3-14-gd3e562907
```
I have the latest version 1.0.4 for ESP32 Dev Module. The version of the UniversalTelegramBot library is the newest downloaded from github.

I created this demo-sketch without the two lines mentioned above:
```
/*******************************************************************
 *  An example of how to use a custom reply keyboard markup.       *
 *                                                                 *
 *                                                                 *
 *  written by Brian Lough                                         *
 *******************************************************************/
#include <WiFi.h>
//#include <WiFiClient.h>
#include <WiFiClientSecure.h>
//#include <UniversalTelegramBot.h>

// Wifi network station credentials
#define WIFI_SSID "xxxx"
#define WIFI_PASSWORD "yyyy"
// Telegram BOT Token (Get from Botfather)
#define BOT_TOKEN "xxxx:yyyy"

const unsigned long BOT_MTBS = 1000; // mean time between scan messages
const int ledPin = 22;

//WiFiClient secured_client;
WiFiClientSecure secured_client;
//UniversalTelegramBot bot(BOT_TOKEN, secured_client);
unsigned long bot_lasttime;          // last time messages' scan has been done
int ledStatus = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.println("Using ESP object:");
  Serial.println(ESP.getSdkVersion());
 
  Serial.println("Using lower level function:");
  Serial.println(esp_get_idf_version());

  // attempt to connect to Wifi network:
  Serial.print("Connecting to Wifi SSID ");
  Serial.print(WIFI_SSID);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  //secured_client.setCACert(TELEGRAM_CERTIFICATE_ROOT); // Add root certificate for api.telegram.org
  Serial.print("\nWiFi connected. IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("Retrieving time: ");
  configTime(0, 0, "pool.ntp.org"); // get UTC time via NTP
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
  Serial.println(now);

  pinMode(ledPin, OUTPUT); // initialize digital ledPin as an output.
  delay(10);
  digitalWrite(ledPin, HIGH); // initialize pin as off
}

void loop()
{
delay(1000);
}
```

Result: The ESP32 immediately connects to the WiFi. Wow, you had a good sense for that change.

After that, I pulled the newest version of UniversalTelegramBot from github and tried the EchoBot example. The result is, that the ESP32 now connects o WiFi, but the bot does not echo the text sent to it.

So far I am a step closer :)

I'll close this issue, since the ESP32 now connects to WiFi. Thank you for your help and have a nice day!

closed 2020-10-24 by gon0