witnessmenow/Universal-Arduino-Telegram-Bot #212: Problems with Json Library and ESP8266 Board Module

DERPNYS, 2020-12-06:
Hi! I'm experiencing some problems trying to make it work. I'm using ArduinoJson 5.15.3 and ESP8266 Module 2.4.2 after reading that these might work but not for me.

The code is: 

`
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

// Inicializamos la conexion WIFI con el Router
char ssid[] = "";     // el nombre de tu Red
char password[] = ""; // la clave de tu Red

// Initialize Telegram BOT
#define BOTtoken ""  // el token de tu BOT, lo obtenemos de BotFather

WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 50; //Tiempo medio entre el escaneo de mensajes
long Bot_lasttime;   //exploracion de el ultimo mensaje
bool Start = false;
String chat_id2;
bool flag = false;
void handleNewMessages(int numNewMessages) {
  Serial.println("handleNewMessages");
  Serial.println(String(numNewMessages));

  for (int i = 0; i < numNewMessages; i++) {
    String chat_id = String(bot.messages[i].chat_id);
    chat_id2 = chat_id;
    String text = bot.messages[i].text;

    String from_name = bot.messages[i].from_name;
    if (from_name == "") from_name = "Guest";

    if (text == "/start") {
      String welcome = "Bienvenido a el BOT de Arduino Center, " + from_name + ".\n";
      welcome += "Escribe /opciones si quieres ver lo que puedo hacer.\n\n";
      //welcome += "/send_test_action : to send test chat action message\n";
      bot.sendMessage(chat_id, welcome);
    }

    if (text == "/opciones") {
      String keyboardJson = "[[\"/ledOn\", \"/ledOff\"],[\"/sensor\"]]";
      bot.sendMessageWithReplyKeyboard(chat_id, "Selecciona una de las siguientes opciones:", "", keyboardJson, true);
    }

    if (text == "/ledOn") {
      digitalWrite(5, HIGH);
      bot.sendMessage(chat_id, "Led Encendido!");
    }

    if (text == "/ledOff") {
      digitalWrite(5, LOW);
      bot.sendMessage(chat_id, "Led Apagado!");
    }

    if (text == "/sensor") {
      if (digitalRead(4) == 0) {
        bot.sendMessage(chat_id, "Sensor en Reposo!");
      } else {
        bot.sendMessage(chat_id, "Sensor Activado!!");
      }
    }
  }
}

void setup() {
  Serial.begin(9600);
[... 259 lines cut ...]
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:786:43: error: 'doc' was not declared in this scope
   DynamicJsonDocument doc(maxMessageLength);
                                           ^
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:787:24: error: 'error' was not declared in this scope
   DeserializationError error = deserializeJson(doc, ZERO_COPY(response));
                        ^
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:794:49: error: expected primary-expression before 'long'
       file_size = doc["result"]["file_size"].as<long>();
                                                 ^
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:794:49: error: expected ';' before 'long'
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp: In member function 'bool UniversalTelegramBot::answerCallbackQuery(const String&, const String&, bool, const String&, int)':
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:802:47: error: 'payload' was not declared in this scope
   DynamicJsonDocument payload(maxMessageLength);
                                               ^
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:811:93: error: expected primary-expression before '>' token
   String response = sendPostToTelegram(BOT_CMD("answerCallbackQuery"), payload.as<JsonObject>());
                                                                                             ^
/Users/X/Documents/Arduino/libraries/Universal-Arduino-Telegram-Bot-master/src/UniversalTelegramBot.cpp:811:95: error: expected primary-expression before ')' token
   String response = sendPostToTelegram(BOT_CMD("answerCallbackQuery"), payload.as<JsonObject>());
                                                                                               ^
exit status 1
Error compilando para la tarjeta Generic ESP8266 Module.
`

Has someone experienced that before? If so I'd be so thankful if anyone can help me. Thanks!!

witnessmenow, 2020-12-07:
> return sendPostMessage(payload.as(), message_id); // if message id == 0

The problem is the error message, you need to have version 6 of Arduino json

measureJson is a function from ArduinoJson 6 but version 5 is installed.

DERPNYS, 2020-12-07:
@witnessmenow Damn... Totally worked, 10/10 haha! Thanks Brian!! Also in case someone else has some similar problems, I ended up using ESP8266 2.7.4 and ArduinoJson as Brian told me. Thanks again!

closed 2020-12-07 by DERPNYS