witnessmenow/Universal-Arduino-Telegram-Bot #213: Using WifiManager

iGodPhil, 2020-12-07:
Hi dudes. 
We have problems with our connection if we use the WifiManager. So is it possible to use both of them in one sketch?
Maybe its helpful to post our sketch.

```
/****************************************************************************************************************************
Autor: Philipp Mielke, Paul Eichner, Joe Fierek, Svenja Meißner
Projekt: smarter Briefkasten
Datum: 21.11.2020
*****************************************************************************************************************************/

// Use from 0 to 4. Higher number, more debugging messages and memory usage.
#define _WIFIMGR_LOGLEVEL_    3

#include <FS.h>

#include <ArduinoJson.h> 

#include <esp_wifi.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

#include <WiFiMulti.h>
WiFiMulti wifiMulti;

  #define USE_SPIFFS      true

  #if USE_SPIFFS
    #include <SPIFFS.h>
    FS* filesystem =      &SPIFFS;
    #define FileFS        SPIFFS
    #define FS_Name       "SPIFFS"
  #else
    // Use FFat
    #include <FFat.h>
    FS* filesystem =      &FFat;
    #define FileFS        FFat
    #define FS_Name       "FFat"
  #endif

  #define ESP_getChipId()   ((uint32_t)ESP.getEfuseMac())

  #define LED_ON            HIGH
  #define LED_OFF           LOW

#define LED_BUILTIN       1         // Pin D1 mapped to pin GPIO1/ADC12 of ESP32, control on-board LED
#define PIN_LED           1         // Pin D1 mapped to pin GPIO1/ADC12 of ESP32, control on-board LED
#define REED              34        // Pin D34 mapped to reed contact
#define WifiReset         35        // Pin D35 mapped to wifi/reset button
#define RedLed            32        // Pin D32 mapped to the red led
#define GreenLed          33        // Pin D33 mapped to the green led
#define mosi              23        // Pin D23 mapped to mosi from rfid
#define rst               22        // Pin D22 mapped to reset from rfid
#define sda               21        // Pin D21 mapped to sda from rfid
#define miso              19        // Pin D19 mapped to miso from rfid
#define sck               18        // Pin D18 mapped to sck from rfid

#define PIN_D0            0         // Pin D0 mapped to pin GPIO0/BOOT/ADC11/TOUCH1 of ESP32
#define PIN_D1            1         // Pin D1 mapped to pin GPIO1/TX0 of ESP32
#define PIN_D2            2         // Pin D2 mapped to pin GPIO2/ADC12/TOUCH2 of ESP32
#define PIN_D3            3         // Pin D3 mapped to pin GPIO3/RX0 of ESP32
#define PIN_D4            4         // Pin D4 mapped to pin GPIO4/ADC10/TOUCH0 of ESP32
#define PIN_D5            5         // Pin D5 mapped to pin GPIO5/SPISS/VSPI_SS of ESP32
#define PIN_D6            6         // Pin D6 mapped to pin GPIO6/FLASH_SCK of ESP32
#define PIN_D7            7         // Pin D7 mapped to pin GPIO7/FLASH_D0 of ESP32
#define PIN_D8            8         // Pin D8 mapped to pin GPIO8/FLASH_D1 of ESP32
#define PIN_D9            9         // Pin D9 mapped to pin GPIO9/FLASH_D2 of ESP32

[... 819 lines cut ...]
            if ( (String(WM_config.WiFi_Creds[i].wifi_ssid) != "") && (strlen(WM_config.WiFi_Creds[i].wifi_pw) >= MIN_AP_PASSWORD_SIZE) )
            {
              LOGERROR3(F("* Add SSID = "), WM_config.WiFi_Creds[i].wifi_ssid, F(", PW = "), WM_config.WiFi_Creds[i].wifi_pw );
              wifiMulti.addAP(WM_config.WiFi_Creds[i].wifi_ssid, WM_config.WiFi_Creds[i].wifi_pw);
            }
          }
        
          saveConfigData();
        }
    
        // Getting posted form values and overriding local variables parameters
        // Config file is written regardless the connection state
        strcpy(PortalSSID, p_PortalSSID.getValue());
        strcpy(PortalPassword, p_PortalPassword.getValue());
    
        // Writing JSON config file to flash for next boot
        writeConfigFile();
    
        digitalWrite(PIN_LED, LED_OFF); // Turn LED off as we are not in configuration mode.
        digitalWrite(RedLed, LOW);
      }
  }
}

```

miguerman, 2021-03-03:
I am interested in this topic. I need to use Wifimanager with Universal arduino telegram. Can someone help

Leoruiz197, 2021-03-08:
hello friends i soved this problem using ESP8266

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>

//NTP server
#include <NTPClient.h>
#include <WiFiUdp.h>

//Telegram bot
#include <UniversalTelegramBot.h>
#include <ArduinoJson.h>

#define BOT_TOKEN "YOUR API HERE" //API KEY telegram

//lista de permissao telegram
#define SENDER_ID_COUNT 2
String validSenderIds[SENDER_ID_COUNT] = {"2222222", "1111111"}; 

//variaveis bot telegram
uint32_t lastCheckTime = 0;
const String STATS = "/status";
const String STATUSID = "/statusid";
const String START = "/start";

//configs bot telegram
X509List cert(TELEGRAM_CERTIFICATE_ROOT);
WiFiClientSecure secured_client;
UniversalTelegramBot bot(BOT_TOKEN, secured_client);

//config NTP
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org");

//wifiManager button
#define TRIGGER_PIN 0

//variavel de millis atual
uint32_t now;

void setup() {
  #ifdef DEBUG
   Serial.begin(9600);
  #endif

  sensor.begin(); //inicializacao sensor de temperatura

  //definicao botao do wifiManeger
  pinMode(TRIGGER_PIN, INPUT);

  //definicao de cliente telegram
  secured_client.setTrustAnchors(&cert);

}

void loop() {
  now = millis(); //captura milissegundo atual
  
  if ( digitalRead(TRIGGER_PIN) == LOW ) { //wifiManeger OnDemand
    WiFiManager wifiManager;
    
    if (!wifiManager.startConfigPortal(Dispositivo_ID)){
      printSerial("failed to connect and hit timeout",1);
      delay(3000);
      ESP.reset();
      delay(5000);
    }
    printSerial("connected...yeey :)",1);
    //secured_client.setTrustAnchors(&cert);
    for(int x = 0; x<SENDER_ID_COUNT; x++){ //apos conexao avisa o status do dispositivo via telegram
      handleStatus(validSenderIds[x]);
    }
  }
  
  //loop Internet connected
  if (WiFi.status() == WL_CONNECTED){

    //verifica novas mensagens no telegram
    if (now - timer > intervalo){
      timer = now;
      int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
      handleNewMessages(numNewMessages);
    }
  }
}

iGodPhil, 2021-03-08:
> I am interested in this topic. I need to use Wifimanager with Universal arduino telegram. Can someone help

we also solved our problem. at first its better you use the original wifiManager from tzapu which works very well with universal telegram bot. we used the development branch of wifiManager because we use the ESP32 but there are also examples for the ESP2688.
we also seperated our code in different functions and call them into the loop if we need them. we've coded them in different tabs.

here are some snippets of our code:

Main page:
```
/****************************************************************************************************************************
Autor: Philipp Mielke, Paul Eichner, Joe Fierek, Svenja Meißner
Projekt: smarter Briefkasten
Datum: 07.03.2021
*****************************************************************************************************************************/

//initialisiert das Dateisystem indem die nötigen Bibliotheken eingebunden werden
#include <FS.h>                   //muss als Erstes angegeben werden
#include <SPIFFS.h>
FS* filesystem =      &SPIFFS;

//nötige Bibliotheken der verschiedenen Funktionen
//#include <DNSServer.h>
//#include <WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <WiFi.h>                 //https://github.com/esp8266/Arduino
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h> //https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <MFRC522.h>              //library responsible for communicating with the module RFID-RC522
#include <SPI.h>                  //library responsible for communicating of SPI bus

//hier werden die nötigen Pins definiert
#define LED_BUILTIN       1         // Pin D1 mapped to pin GPIO1/ADC12 of ESP32, control on-board LED
#define schloss           12        // Pin D12 mapped to schloss
#define REED              34        // Pin D34 mapped to reed contact
#define WifiReset         35        // Pin D35 mapped to wifi/reset button
#define RedLed            32        // Pin D32 mapped to the red led
#define GreenLed          33        // Pin D33 mapped to the green led
#define mosi              23        // Pin D23 mapped to mosi from rfid
#define rst               22        // Pin D22 mapped to reset from rfid
#define sda               21        // Pin D21 mapped to sda from rfid
#define miso              19        // Pin D19 mapped to miso from rfid
#define sck               18        // Pin D18 mapped to sck from rfi

//wird für die Authentifizierung verwendet
MFRC522::MIFARE_Key key;

//Authentifizierung die den Statuscode zurückgibt
MFRC522::StatusCode status;

//definiert die Pins für das Modul RC522
MFRC522 mfrc522(sda, rst); 

WiFiClientSecure secured_client;

//Difinition der globen Variablen auf die alle Funktionen zugreifen können
char telegram_chatID[40];
char telegram_token[70];
char schloss_notfall[5] = "nein";
int card_uid[10][4] = {0, 0, 0, 0};

bool wifi_merker = false;                   //Anzeige, dass die Wifi Verbindung verloren wurde
bool telegram_init_fertig = false;          //wird benötigt um den Telegrambot einmalig in der Loopfunktion zu Initialisieren

unsigned long timer_wifi_start = 0;         //Zeit um die rote Led zu toggeln bei Wlan Ausfall während des Betriebs
bool timer_wifi_laeuft = false;             //true wenn noch auf die Verzögerung gewartet wird

bool daten_speichern = false;               //flag um die Daten zu speichern
bool wifi_verbindung_verloren = false;      //Flag um den User zu benachrichtigen, dass die Wifi Verbindung verloren wurde

[... 423 lines cut ...]
  //Serial.println("Retrieving time: ");
  configTime(0, 0, "pool.ntp.org");                    
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    //Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
  //Serial.println(now);

  //jetzt ist der Bot online
  if(!wifi_verbindung_verloren)
    telegramMessage("Der Briefkasten ist online.");
  else
  {
    telegramMessage("Der Briefkasten hat zwischenzeitlich die Internetverbindung verloren und ist jetzt wieder online.\nBitte überprüfe, ob Post angekommen ist.");
    wifi_verbindung_verloren = false;
  }
  telegram_init_fertig = true;
}
```

You see, after any new wifi connection we initialize the telegram bot and send our messages with this function:
`telegramMessage("string to send the message you want");`

closed 2021-03-08 by iGodPhil