witnessmenow/Universal-Arduino-Telegram-Bot #215: error: 'X509List' does not name a type

ant1fr1z, 2020-12-10:
sketch_telegram_echo_bot:29:1: error: 'X509List' does not name a type
 X509List cert("YOURPUBLIC.pem");
 ^
D:\Documents\Arduino\sketch_telegram_echo_bot\sketch_telegram_echo_bot.ino: In function 'void setup()':
sketch_telegram_echo_bot:51:18: error: 'class axTLS::WiFiClientSecure' has no member named 'setTrustAnchors'
   secured_client.setTrustAnchors(&cert); // Add root certificate for api.telegram.org
                  ^
sketch_telegram_echo_bot:51:35: error: 'cert' was not declared in this scope
   secured_client.setTrustAnchors(&cert); // Add root certificate for api.telegram.org
                                   ^
sketch_telegram_echo_bot:63:28: error: 'time' was not declared in this scope
   time_t now = time(nullptr);
                            ^
exit status 1
'X509List' does not name a type

saikek, 2021-03-19:
Are you using platformio ? What are your dependencies ? What's your board ?

yeseswisree, 2021-05-20:
> sketch_telegram_echo_bot:29:1: error: 'X509List' does not name a type
> X509List cert("YOURPUBLIC.pem");
> ^
> D:\Documents\Arduino\sketch_telegram_echo_bot\sketch_telegram_echo_bot.ino: In function 'void [...]

even got the same error.. can anyone help me?

yeseswisree, 2021-05-20:
'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope
this is my error. can anyone help me?

saikek, 2021-05-20:
Any details ? 
Are you using Arduino or VS Code + platformio.
What version of library are you using ?

heiko178, 2021-06-01:
i have the same problem. 
VS Code + platformio

Shoxrux-96, 2021-06-09:
Arduino: 1.8.15 Hourly Build 2021/05/31 10:33 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\User\Documents\Arduino\libraries\UniversalTelegramBot\src/UniversalTelegramBot.h:31:0,

                 from C:\Users\User\Documents\Arduino\ChangeUID\telegram-esp_bot\telegram-esp_bot.ino:2:

C:\Users\User\Documents\Arduino\libraries\UniversalTelegramBot\src/TelegramCertificate.h:9:38: error: redefinition of 'const char TELEGRAM_CERTIFICATE_ROOT []'

 const char TELEGRAM_CERTIFICATE_ROOT[] = R"=EOF=(
                                      ^
In file included from C:\Users\User\Documents\Arduino\ChangeUID\telegram-esp_bot\telegram-esp_bot.ino:1:0:

C:\Users\User\Documents\Arduino\libraries\UniversalTelegramBot\src/TelegramCertificate.h:9:12: error: 'const char TELEGRAM_CERTIFICATE_ROOT [1369]' previously defined here

 const char TELEGRAM_CERTIFICATE_ROOT[] = R"=EOF=(
            ^
telegram-esp_bot:25:3: error: 'X509List' does not name a type

   X509List cert(TELEGRAM_CERTIFICATE_ROOT);
   ^
C:\Users\User\Documents\Arduino\ChangeUID\telegram-esp_bot\telegram-esp_bot.ino: In function 'void setup()':

telegram-esp_bot:94:12: error: 'class WiFiClientSecure' has no member named 'setTrustAnchors'

     client.setTrustAnchors(&cert); // Add root certificate for api.telegram.org
            ^
telegram-esp_bot:94:29: error: 'cert' was not declared in this scope

     client.setTrustAnchors(&cert); // Add root certificate for api.telegram.org

exit status 1

'X509List' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
//////////////////////////////////////////////////////////////////
I also have a similar problem if anyone knows, if anyone knows, help.

lucasquintanilla, 2021-06-18:
Some news about this issue?

wayn-suta, 2021-07-04:
I've had the same problem. then I did some updates as below, and it works!!
library:
Universal Telegram Bot v. 1.3.0
ArduinoJson 6.15.2

borads:
esp 8288 2.5.1

wayn-suta, 2021-07-04:
> Some news about this issue?

've had the same problem. then I did some updates as below, and it works!!
library:
Universal Telegram Bot v. 1.3.0
ArduinoJson 6.15.2

boards:
esp 8288 2.5.1

hope this can help u

ProtonMeow, 2022-02-22:
Arduino: 1.8.20 Hourly Build 2021/12/20 07:33 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Pyton_to_Arduino:46:15: error: 'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

   46 | X509List cert(TELEGRAM_CERTIFICATE_ROOT);

      |               ^~~~~~~~~~~~~~~~~~~~~~~~~

Multiple libraries were found for "UniversalTelegramBot.h"

 Used: C:\Users\User\Documents\Arduino\libraries\UniversalTelegramBot

 Not used: C:\Users\User\Documents\Arduino\libraries\Universal-Arduino-Telegram-Bot-master

exit status 1

'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

This is my issues please help me