witnessmenow/Universal-Arduino-Telegram-Bot #216: bot.getUpdates takes long time

Tantrumm, 2020-12-20:
Hello! I tested the TelegramBot library and noticed that the next line takes about 3.5 seconds. It's very long.

`int numNewMessages = bot.getUpdates(bot.last_message_received + 1);`

(It can be reduced by reducing `bot.waitForResponse` to **200**, but the best result is 1.5 second. It's also too long.

Who is faces with this before, why it's so long delay? Can it be fixed somehow?

RomeHein, 2021-02-20:
Yeah same here. The best way would be to use websockets, but that would mean an asyncTCP library that would handle https...

mgx0, 2021-03-02:
Setting connection to insecure would help in my opinion. All I care about is encrypted communication. Having correct time set, verified certificate is nice but not for this cpu and for this purpose. 

Try insecure client but still https