witnessmenow/Universal-Arduino-Telegram-Bot #219: Telegram and ESPNOW problems

thefly0, 2020-12-24:
Hello,
can anyone help in suggesting a code, if possibile, to run simultaneously ESPNOW (to receive data periodically from my sensor) and telegram to read data from internet. I'm using and ESP32D (node32s). At the moment the only solution I found is disconnecting WiFi when not checking Telegram bot.  Below I attached my code.
thanks in advance
Stefano

****************************************************
#include <WiFi.h>
#include <esp_now.h>

// aggiunta per telegram
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>
#include <ArduinoJson.h>

// Use @myidbot to find out the chat ID of an individual or a group
// Also note that you need to click "start" on a bot before it can message you
#define CHAT_ID "zzzzzzzz"

// Initialize Telegram BOT
#define BOTtoken "xxxxxxxxxx"  // your Bot Token (Get from Botfather)
#define VCC_CONV 4.14/2.27              // fattore di conversione da (ESP.getVcc() / 1024.0) per ottenere la Vin

WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);

typedef struct struct_message {
  float temp, hum, pres, vcc;
  uint8_t sens_id;
} struct_message;

struct_message myData;

const char* ssid = "myssid";
const char* password= "mypsw";

long timestamp_telegram = 0;

void setup() {

  Serial.begin(115200);

#ifdef ESP8266
  Serial.println("Running on ESP8266");
#elif ESP32
  Serial.println("Running on ESP32");
#endif

  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);

  WiFi.mode(WIFI_STA);

  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  } else {
    Serial.println("ESPNOW initialized!");
  }

  //connect to WiFi AP
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while ((WiFi.status() != WL_CONNECTED)) {
    delay(500);
    Serial.print(".");
  }
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("\nConnected to " + String(ssid) + "(" + String(WiFi.RSSI()) + " dBm)");
  } else {
    Serial.println("\nNot connected to " + String(ssid));
  }

  // Once ESPNow is successfully Init, we will register for recv CB to
  // get recv packer info
  esp_now_register_recv_cb(OnDataRecv);

  bot.sendMessage(CHAT_ID, "Welcome back....", "");
  Serial.println("Setup finished!");
}

void loop() {
  if ((millis() - timestamp_telegram) > 10000u) {
    WiFi_connect();
    timestamp_telegram = millis();
    // verifica le interrogazioni da telegram
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    Serial.println("Check BOT: newMsg=" + String(numNewMessages));
    while (numNewMessages) {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }
    WiFi_disconnect();
  }
}

// callback function that will be executed when data is received
void OnDataRecv(const uint8_t * mac, const uint8_t *incomingData, int len) {
  memcpy(&myData, incomingData, sizeof(myData));

  if (myData.sens_id == 1) {
    Serial.println("#1:T=" + String(myData.temp, 1) + "C ; H=" + String(myData.hum, 0) + "% ; Vcc=" + String(myData.vcc) + "V");
  }
  if (myData.sens_id == 2) {
    Serial.println("#2:T=" + String(myData.temp, 1) + "C ; H=" + String(myData.hum, 0) + "% ; Vcc=" + String(myData.vcc * VCC_CONV, 2) + "V");
  }
}

String get_meteo_data(void) {
  String str;
  str = "Temp in = " + String(myData.temp, 1) + " C\n";
  str += "Humi in = " + String(myData.hum, 0) + " %\n";
  str += "Vcc out = " + String(VCC_CONV * myData.vcc, 2) + " V\n";
  return str;
}

//Handle what happens when you receive new messages
void handleNewMessages(int numNewMessages) {
  String readings;
  Serial.println("handleNewMessages");
  Serial.println(String(numNewMessages));

  for (int i = 0; i < numNewMessages; i++) {
    // Chat id of the requester
    String chat_id = String(bot.messages[i].chat_id);
    if (chat_id != CHAT_ID) {
      bot.sendMessage(chat_id, "Unauthorized user", "");
      continue;
    }
    // Print the received message
    String text = bot.messages[i].text;
    Serial.println(text);
    String from_name = bot.messages[i].from_name;

    if (text == "/start") {
      String welcome = "Welcome, " + from_name + ".\n";
      welcome += "Use the following command to get current readings.\n\n";
      welcome += "/readings \n";
      welcome += "/options \n";
      bot.sendMessage(chat_id, welcome, "");
    }

    if (text == "/readings") {
      readings = get_meteo_data();
      bot.sendMessage(chat_id, readings, "");
    }

    if (text == "/options")
    {
      String keyboardJson = "[[\"/readings\", \"/options\"],[\"/start\"]]";
      bot.sendMessageWithReplyKeyboard(chat_id, "Choose from one of the following options", "", keyboardJson, true);
    }

  }
}

void WiFi_connect(void) {
  Serial.println("CONN->SSID: "+String(WiFi.SSID()));
  WiFi.begin(ssid, password);
  Serial.println("6 : WiFi status = " + String(WiFi.status()));
  while ((WiFi.status() != WL_CONNECTED)) {
    delay(500);
    Serial.print(".");
  }
}
void WiFi_disconnect(void) {
  Serial.println("DIS->SSID: "+String(WiFi.SSID()));
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  WiFi.mode(WIFI_STA);
}

thefly0, 2020-12-24:
> Hello,
> can anyone help in suggesting a code, if possibile, to run simultaneously ESPNOW (to receive data periodically from my sensor) and telegram to read data from internet. I'm using and ESP32D [...]

alvaroluiz2, 2021-12-20:
Please request help.
- it would be possible to use the telegram bot using esp8266 and espnow. Idea to make a small network (one room) with ESPNOW and interconnect in a corridor to other rooms, via RS485, a small central ESP8266 via Telegram Bot.
Thanks
AlvaroLuiz2@yahoo.com.br