witnessmenow/Universal-Arduino-Telegram-Bot #220: Send HIGH quality photo from ESP32-CAM

array81, 2020-12-25:
I tried official ESP32CAM example.
This work only if framesize is set to FRAMESIZE_QVGA, in alternative I don't get error but message with image is not send.
Try to change FRAMESIZE_QVGA to FRAMESIZE_VGA or any other resoltution.

Is there a way to send image with a big resolution?

espadrift, 2021-01-10:
@array81  what example are you using? I found many use uint_16 types for the image buffer sizes. These need to be changed to uint_32.

array81, 2021-01-10:
@espadrift the official ESP32CAM example, you can find it in this repository.

anshchawla521, 2021-02-17:
i am also trying to to the same thing but the code just gets stuck when we change the resolution

diibify, 2021-03-01:
I have a very similar problem. Many (all other) ESP32cam foto examples do work (CameraWebServer etc). In this example the flashing does work and the interaction in total, too. But the /photo action doesnt. I see in Serial Monitor that he is probably sending the photo and is also saying that it was successfull. But there is no message in telegram. Not a empty message - none. I also tried the hint in the first post changing the FRAMESIZE_UXGA to FRAMESIZE_QVGA. Even with other UniversalTelegram Bot examples I get no message in telegram. I tried very much the other days - I do not have an idea yet. I would be very happy if there is a tip from you. 
[Git.zip](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/files/6060496/Git.zip)

DaniDerBaer, 2021-03-06:
@diibify I have exactly the same problem. In the Serial Monitor it says that it was successful, but there isn't a message in telegram.

FBMinis, 2021-03-15:
After experiencing dificulties sending large resolution photos, I found an application that works. I have two of these units sending a photo every two minutes during the day to my bot, resolution is XGA:

https://github.com/robotzero1/esp32cam-telegram (thanks to robotzero1 for his work)
[image]

philicibine, 2021-07-04:
This seems to work for me.  Taken from robotzero1's example and used with this library.

`bool isMoreDataAvailable() {
  return (fb_length - currentByte);
}

uint8_t photoNextByte() {
  currentByte++;
  return (fb_buffer[currentByte - 1]);
}

void take_send_photo() {
  camera_fb_t * fb = NULL;
  fb = esp_camera_fb_get();
  currentByte = 0;
  fb_length = fb->len;
  fb_buffer = fb->buf;

  // send the photo to the private channel
  myBot.sendPhotoByBinary(group_id, "image/jpeg", fb->len, isMoreDataAvailable, photoNextByte, nullptr, nullptr);

  esp_camera_fb_return(fb);
  fb_length = NULL;
  fb_buffer = NULL;

}`