witnessmenow/Universal-Arduino-Telegram-Bot #228: Telegram not working on Arduino core for the ESP32

turbonatta, 2021-03-01:
With the latest core 1.0.5 released on Febraruary 23rd 2021 Telegram no longer work.
No error message during compilation on Arduino IDE.
The application compiled with the newest core does not send or received any message on Telegram platform.
Reverting to previous 1.0.4 core fix the problem.
Any suggestion?

mgx0, 2021-03-02:
looks like none of them work. I have problems connecting to HTTPS so this has to be somewhere level below

turbonatta, 2021-03-02:
I posted an issues on espressif/esp32 repository about this problem but it was closed immediately with comment:
"please take this issue to the library maintainer ;) we do not support external libraries in this issue tracker"

witnessmenow, 2021-03-02:
Can you enable debug logs and see where it fails to connect?

mgx0, 2021-03-02:
for me it works in PlatformIO. I don't have much time to update it again but I saw the problem was in connecting HTTPS, simply returned nothing. When I have time I'll update and post the results but I'll leave it to @turbonatta turbonatta for now

This is what works for me now

UniversalTelegramBot 1.3.0
ArduinoJson 6.17.3
Espressif 8286 2.6.2 platform

turbonatta, 2021-03-03:
I developed the applications on Arduino IDE. As you know debug capability are very limited. I'll try this evening (Italian time). I suspect maybe the problem is in WiFiClientSecure

mgx0, 2021-03-03:
> I developed the applications on Arduino IDE. As you know debug capability are very limited. I'll try this evening (Italian time). I suspect maybe the problem is in WiFiClientSecure

I think the request was to enable debug. 

Uncomment line 25 to look like this:
#define TELEGRAM_DEBUG 1

Run your program again And post the serial output here.

turbonatta, 2021-03-03:
Sorry, I misandurstood, :)

turbonatta, 2021-03-03:
Sorry, I misunderstood, :):)

witnessmenow, 2021-03-03:
Enabling Debug on the library as @mgx0 mentioned is a good start. if it fails at the connect stage we can generate a sample sketch to go back to the Espressif guys with

turbonatta, 2021-03-03:
This is the serial output with debug enable:
19:41:38.647 -> 
19:41:38.647 -> ESP IP Address: http://192.168.1.194
19:41:40.584 -> sendPostMessage: SEND Post Message: {"chat_id":"818311005","text":"Programma caricato"}
19:41:40.584 -> [BOT Client]Connecting to server
19:41:40.584 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.631 -> [BOT Client]Conection error
19:41:40.631 -> 
19:41:40.631 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.677 -> 
19:41:40.677 -> [BOT Client]Connecting to server
19:41:40.677 -> [BOT Client]Conection error
19:41:40.724 -> 
19:41:40.724 -> [BOT Client]Connecting to server
19:41:40.724 -> [BOT Client]Conection error
19:41:40.724 -> 
19:41:40.724 -> [BOT Client]Connecting to server
19:41:40.724 -> [BOT Client]Conection error
19:41:40.724 -> 
19:41:40.724 -> [BOT Client]Connecting to server
19:41:40.724 -> [BOT Client]Conection error
19:41:40.724 -> 
19:41:40.724 -> [BOT Client]Connecting to server
19:41:40.724 -> [BOT Client]Conection error
19:41:40.724 -> 
[... 4191 lines cut ...]
19:41:48.728 -> Wednesday, March 03 2021 19:41:49
19:41:48.728 -> Day of week: Wednesday
19:41:48.728 -> Month: March
19:41:48.728 -> Day of Month: 03
19:41:48.728 -> Year: 2021
19:41:48.728 -> Hour: 19
19:41:48.728 -> Minute: 41
19:41:48.728 -> Second: 49
19:41:48.728 -> 
19:41:58.041 -> GET Update Messages
19:41:58.041 -> [BOT]Connecting to server
19:41:58.041 -> [BOT]Conection error
19:41:58.041 -> Received empty string in response!
19:42:18.027 -> GET Update Messages
19:42:18.074 -> [BOT]Connecting to server
19:42:18.074 -> [BOT]Conection error
19:42:18.074 -> Received empty string in response!
19:42:38.043 -> GET Update Messages
19:42:38.043 -> [BOT]Connecting to server
19:42:38.043 -> [BOT]Conection error
19:42:38.043 -> Received empty string in response!
19:42:58.028 -> GET Update Messages
19:42:58.028 -> [BOT]Connecting to server
19:42:58.075 -> [BOT]Conection error
19:42:58.075 -> Received empty string in response!

turbonatta, 2021-03-03:
I am running a simple web server as well to display few sensor data and it works without problem

turbonatta, 2021-03-03:
Reinstalling previous 1.0.4 core library ESP32:

20:15:38.780 -> ets Jun  8 2016 00:22:57
20:15:38.780 -> 
20:15:38.780 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:15:38.780 -> configsip: 0, SPIWP:0xee
20:15:38.780 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:15:38.814 -> mode:DIO, clock div:1
20:15:38.814 -> load:0x3fff0018,len:4
20:15:38.814 -> load:0x3fff001c,len:1216
20:15:38.814 -> ho 0 tail 12 room 4
20:15:38.814 -> load:0x40078000,len:9720
20:15:38.814 -> ho 0 tail 12 room 4
20:15:38.814 -> load:0x40080400,len:6352
20:15:38.814 -> entry 0x400806b8
20:15:39.562 -> 
20:15:39.562 -> ESP IP Address: http://192.168.1.194
20:15:41.569 -> sendPostMessage: SEND Post Message: {"chat_id":"818311005","text":"Programma caricato"}
20:15:41.603 -> [BOT Client]Connecting to server
20:15:43.092 -> Posting:{"chat_id":"818311005","text":"Programma caricato"}
20:15:43.261 -> 
20:15:43.261 -> {"ok":true,"result":{"message_id":1564,"from":{"id":1345783975,"is_bot":true,"first_name":"Turbonatta","username":"Turbonatta_bot"},"chat":{"id":818311005,"first_name":"Carlo","last_name":"Casini","type":"private"},"date":1614798944,"text":"Programma caricato"}}
20:15:43.261 -> 
20:15:43.261 -> {"ok":true,"result":{"message_id":1564,"from":{"id":1345783975,"is_bot":true,"first_name":"Turbonatta","username":"Turbonatta_bot"},"chat":{"id":818311005,"first_name":"Carlo","last_name":"Casini","type":"private"},"date":1614798944,"text":"Programma caricato"}}
20:15:43.295 -> Closing client
20:15:43.329 -> Wednesday, March 03 2021 20:15:44
20:15:43.363 -> Day of week: Wednesday
20:15:43.363 -> Month: March
20:15:43.363 -> Day of Month: 03
20:15:43.363 -> Year: 2021
20:15:43.363 -> Hour: 20
20:15:43.363 -> Minute: 15
20:15:43.363 -> Second: 44
20:15:43.363 -> 
20:15:44.117 -> Temperature above threshold. Current temperature: 18.22C
20:15:59.140 -> GET Update Messages
20:15:59.140 -> [BOT]Connecting to server
20:16:00.262 -> sending: bot1345783975:AAHVWcKHXTTiHlvbNHXu-ljJhRvowhV_n0g/getUpdates?offset=1&limit=1
20:16:00.331 -> 
20:16:00.331 -> {"ok":true,"result":[]}
20:16:00.331 -> 
20:16:00.331 -> incoming message length 23
20:16:00.331 -> Creating DynamicJsonBuffer
20:16:00.331 -> GetUpdates parsed jsonObj: {"ok":true,"result":[]}
20:16:00.363 -> no new messages
20:16:00.363 -> Closing client

mgx0, 2021-03-03:
That did not say much about the problem. I'll install arduino and try it myself in an hour

turbonatta, 2021-03-03:
Thanx mgx0! 
I got the same result.
I am using AsyncWebServer library for my simple server which it seems not effected by the new core . 
But since I need the Telegram features for the moment I'll keep using 1.0.4
I am not sure what is going to be the next step since I was already turn down when I reported the problem on expressif repository.

mgx0, 2021-03-04:
They cannot turn you down with this above. This has nothing to do with external library. Reopen the issue there and post the link here. Or maybe @witnessmenow could help, I'm not very github-experienced

witnessmenow, 2021-03-04:
I think that example is great, I would maybe use something that doesn't crash 1.4 though 😅

If I get some time over the next few days I can raise the issue, but I'm really swamped at the moment so I would really appreciate if someone else could take charge of it!

turbonatta, 2021-03-04:
I am going to raise the issue, this evening as soon I'll get home, For whatevere reason when I run the test code on 1.0.4 core it did not crush. I got a "404" reply plus some other information from Google and the "-1" error code on 1.0.5 core as seen by mgx0

mgx0, 2021-03-04:
it crashes because of the length of the response.

give it https://www.google.com/asdf to make sure you get 404

or you can use for example https://blockchain.info/tobtc?currency=USD&value=500 as the URL but you have to replace the certificate in the sketch

turbonatta, 2021-03-04:
I raised the issue on espressif repository posting the output of the sample program with the two cores and it was closed right away.
They reply by advising to run on "Core level debug". With that turn on the reason of the fail connection is:

23:10:29.272 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
23:10:29.272 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
23:10:29.339 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
23:10:29.339 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
23:10:29.339 -> [D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.98, MASK: 255.255.255.0, GW: 192.168.1.1
23:10:30.290 -> Connecting to WiFi..
23:10:30.290 -> Connected to the WiFi network
23:10:30.290 -> [D][HTTPClient.cpp:287] beginInternal(): host: www.google.com port: 443 url: /adsf
23:10:30.290 -> [D][HTTPClient.cpp:573] sendRequest(): request type: 'GET' redirCount: 0
23:10:30.290 -> 
23:10:30.461 -> [E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
23:10:30.461 -> [E][WiFiClientSecure.cpp:127] connect(): start_ssl_client: -9984
23:10:30.461 -> [D][HTTPClient.cpp:1112] connect(): failed connect to www.google.com:443
23:10:30.495 -> [W][HTTPClient.cpp:1411] returnError(): error(-1): connection refused
23:10:30.495 -> Error on HTTP request-1
23:10:30.495 -> [D][HTTPClient.cpp:394] disconnect(): tcp is closed
23:10:30.495 -> 

I am not sure what to make but it looks like that the certificate is checked before it try to connect.
Any suggestion?

witnessmenow, 2021-03-05:
I just tested a fresh install of the ESP32 Core and the library is working for me. Do you have a different ESP32 you could try @turbonatta ? I have a feather huzzah that is funny with different versions of the ESP32 core and im not sure why.

[image]

turbonatta, 2021-03-05:
I have two esp32-wroom-devkitc v2 and they both shows the same behaviour

turbonatta, 2021-03-05:
I finally figure out the problem! I build my project using an example for Universal-Arduino-Telegram-bot found on internet. This example establish a secure connection without using any root certificate. With ESP32 core 1.0.4 the missing certificate generate only a warning, when "Core Level Debug" is on, to improve security by using a certificate.  With ESP32 core 1.0.5 it refuse the connection unless you provide a certificate or you execute client.SetInsicure(); I test both solution and the Telegram interface works fine with the same warning to improve security when using SetInsicure();
Thanks everybody!

marmuck, 2021-03-05:
man, thanks so much! I just spent the whole day picking my project apart as I thought I must have broken something – that one line fixed it...

closed 2021-03-06 by turbonatta

Lau-res, 2021-06-17:
Hi! What versions of ArduinoJson, UniversalTelegramBot , esp32 core, WiFiClientSecure and Arduino IDE are you using. I have the same error but _client.setInsecure_ and _secured_client.setCACert(TELEGRAM_CERTIFICATE_ROOT);_ don't work for me.
¿Could someone tell me what libraries work fine, please?
Thanks a lot

Flavelius, 2022-03-04:
i would like to know the same, both solutions also don't work for me