witnessmenow/Universal-Arduino-Telegram-Bot #231: Connection error after ArduinoJson upgrade

array81, 2021-03-21:
After upgrade ArduinoJson to 6.17.3 I get this error when try to use the library:
[BOT Client]Connecting to server
[BOT Client]Conection error

Lau-res, 2021-06-17:
Didi you solve it? I have a similar libraries incompatibility issue and the connection between esp32 and telegram bot doesn't work.
What versions work for you? Thanks a lot

witnessmenow, 2021-06-17:
What code are you using?

Esp32 now forces a https check unless you tell it not too, I believe all the recent examples of this library use the cert.

Lau-res, 2021-06-17:
How do you tell it not to do de https check?
I'm trying with EchoBot example from Universal Telegram Bot and the connection doesn't work.

witnessmenow, 2021-06-17:
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP32/EchoBot/EchoBot.ino#L55 Should cover it.

But you could replace this this with: `secured_client.setInsecure();` to try ignoring the HTTPS check altogether

Lau-res, 2021-06-17:
I continue with this error [BOT Client]Connecting to server [BOT Client]Conection error.
What versions of Arduino IDE, UniversalTelegramBot, ArduinoJson and esp32 did you use?
Thanks a lot!

paulmarcelflo, 2021-11-01:
I have the same error : 

**_GET Update Messages
[BOT]Connecting to server
[BOT]Conection error
Received empty string in response!_**

Arduino IDE 1.8.16
Esp32 1.0.6
ArduinoJson 6.18.5

Has anybody solved this issue ?