witnessmenow/Universal-Arduino-Telegram-Bot #232: Can't use the Bulk Messages for ESP8266  code!

Clarathms, 2021-03-23:
I copied the Bulk Message code the same way it was written, and added the SSID ,password and bot token as it told me to do.
But when I tested it, all the bot functions wouldn't work. For example : Even tho there were 2 subscribed users, it just kept saying ´´null´´

I would like to know how to proceed

devinasander, 2021-09-30:
Save file with this mehtod to SPIFFS. Then you can attach the image file to the Telegram message

usage: downloadAndSaveFile("/image.jpg","http://192.168.1.40/jpg/image.jpg");  

--------------
void downloadAndSaveFile(String fileName, String  url){
  
  HTTPClient http;

  Serial.println("[HTTP] begin...\n");
  Serial.println(fileName);
  Serial.println(url);
  http.begin(url);
  Serial.printf("[HTTP] GET...\n", url.c_str());
  // start connection and send HTTP header
  int httpCode = http.GET();
  if(httpCode > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTP] GET... code: %d\n", httpCode);
      Serial.printf("[FILE] open file for writing %d\n", fileName.c_str());
      
      File file = SPIFFS.open(fileName, "a+");

      // file found at server
      if(httpCode == HTTP_CODE_OK) {

          // get lenght of document (is -1 when Server sends no Content-Length header)
          int len = http.getSize();
          // create buffer for read
          uint8_t buff[128] = { 0 };

          // get tcp stream
          WiFiClient * stream = http.getStreamPtr();

          // read all data from server
          while(http.connected() && (len > 0 || len == -1)) {
              // get available data size
              size_t size = stream->available();
              if(size) {
                  // read up to 128 byte
                  int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
                  // write it to Serial
                  //Serial.write(buff, c);
                  file.write(buff, c);
                  if(len > 0) {
                      len -= c;
                  }
              }
              delay(1);
          }

          Serial.println();
          Serial.println("[HTTP] connection closed or file end.\n");
          Serial.println("[FILE] closing file\n");
          file.close();
          
      }
      
  }
  http.end();

  
}

wirewound, 2021-12-04:
I can not solve this in a pretty way but i found that if e new subscribedUsersFile is created `serializeJson(users, subscribedUsersFile)` writes "null" in the file because `getSubscribedUsers()` returns null becaus the file is empty yet.

```
if (usersDoc.isNull())
      {
        users = usersDoc.to<JsonObject>();
      }
      else 
      { 
        users = usersDoc.as<JsonObject>();
      }
```
This was my workaround for the problem.

SebouhArmenakian, 2022-01-01:
wirewound, you saved my days trying to figure out the issue why I always get "null" for "/showallusers".

I added your suggestion in bool addSubscribedUser(String chat_id, String from_name) after serializeJson(users, subscribedUsersFile); and the sketch works like a charm.

Thank you  @witnessmenow Brian Lough and @wirewound 

> I can not solve this in a pretty way but i found that if e new subscribedUsersFile is created `serializeJson(users, subscribedUsersFile)` writes "null" in the file because `getSubscribedUsers()` [...]

SebouhArmenakian, 2022-01-05:
Unfortunately for unforeseen reasons the suggestion of @wirewound doesn't work always.
Based on his suggestions adding the below by replacing between //users.getOrCreateMember(chat_id, from_name); and subscribedUsersFile.close();  will solve the issue:-

if ((usersDoc.isNull()) && (subscribedUsersFile))
  {
     subscribedUsersFile.print("{\"" + chat_id + "\":" + "\"" + from_name + "\"}" );
  }
  else
  {
    serializeJson(users, subscribedUsersFile);
    users = usersDoc.as<JsonObject>();
  }

Memogcia, 2022-08-14:
The good solution is to have the following lines: 

`//users.getOrCreateMember(chat_id, from_name);
  if ((usersDoc.isNull()) && (authorizedUsersFile))
  {
    authorizedUsersFile.print(INITIAL_STRING);
    Serial.println("Created file from string");
  }
  else 
  { 
    users = usersDoc.as<JsonObject>();
    serializeJson(users, authorizedUsersFile);
    Serial.println("Users doc not null");
  }
  authorizedUsersFile.close();
  return true;`

Memogcia, 2022-08-14:
`
if ((usersDoc.isNull()) && (authorizedUsersFile))
  {
    authorizedUsersFile.print(INITIAL_STRING);
    Serial.println("Created file from string");
  }
  else 
  { 
    users = usersDoc.as<JsonObject>();
    serializeJson(users, authorizedUsersFile);
    Serial.println("Users doc not null");
  }
  authorizedUsersFile.close();
  return true;
`