witnessmenow/Universal-Arduino-Telegram-Bot #237: SendMessageFromEvent example returns 0 on NodeMcu V3

FelixBurkhard, 2021-04-04:
Hello everyone,

thanks for this nice library :)
I tried to send a message to my telegrambot using the [SendMessageFromEvent](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP8266/SendMessageFromEvent/SendMessageFromEvent.ino) example. I changed the WiFi SSID and password, and the BotToken and ChatID to the corresponding values from my bot. Unfortunatly, I always get a 0 as return value from `bot.sendMessage(CHAT_ID, "Bot started up", "");` 
I use the NodeMcu V3 ESP8266 board. Has anyone ever experienced a similar problem or has any idea that could help me to solve this? If you need any further information to investigate just let me know.

Thanks in advance.
Felix

iurii-untilov, 2021-04-25:
Hi @FelixBurkhard ,
I've experienced the same behavior, because I tried to simplify the example and removed some code.
It looks like the code related to time  is mandatory for the setup:
`  
  Serial.print("Retrieving time: ");
  configTime(0, 0, "pool.ntp.org"); // get UTC time via NTP
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
`

After adding this code into setup() function I was able to send messages.
Regards.

FelixBurkhard, 2021-04-27:
Hello @iurii-untilov thanks alot for your response.
Unfortunatly, adding your code snippet did not change the behaviour.
Did you change anything else?
Regards,
Felix

FelixBurkhard, 2021-04-27:
Ok I fixed it.
I had a typo in my BotID :no_mouth:
Thanks again to @iurii-untilov 👍

closed 2021-04-27 by FelixBurkhard