witnessmenow/Universal-Arduino-Telegram-Bot #238: Modulo GPRS  and telegram

raleortiz, 2021-04-10:
I want to be able to send messages through a GPRS module, since instead there is no wifi, I saw a comment that the SIM800L module can no longer be used, that the GPRS module would be the one that allows me to do the project. Can't use WIFI.

SMauser, 2021-05-03:
I´m also very interested in any solution. Have the same question/problem :(

d0ct0rd-dalimo, 2021-05-24:
Hello! I'm using UniversalTelegramBot in pair with ArduinoBearSSL to communicate with api.telegram.org via GSM modem. In bot debug, I see incomplete input in response:
```
Checking bot...
GET Update Messages
[BOT]Connecting to server
1621841932
.... connected to server

{"ok":true,"result":[{"update_id":669498975,
"message":{"message_id":8,"from":{"id":387291095,"is_bot":false,"first_nam

incoming message length 119
Creating DynamicJsonBuffer
GetUpdates parsed jsonDoc: {"ok":true,"result":[{"update_id":669498975,"message":{"message_id":8,"from":{"id":387291095,"is_bot":false}}}]}
Failed to parse update, the message could be too big for the buffer. Error code: IncompleteInput
Closing client
```

I tried to tweak BearSSL and TinyGSM buffers but it's always 119 bytes of response. Or is it some buffer else, or the unexpected end of response...

Here is the usage:
```
TinyGsm modem(SerialAT);
TinyGsmClient client(modem);
BearSSLClient sslClient(client);

UniversalTelegramBot bot(BOTtoken, sslClient);
```

d0ct0rd-dalimo, 2021-09-18:
Issue solved by using ArduinoBearSSL, but you need to increase input and output buffers of ArduinoBearSSL

My buffers are:
```
#ifndef BEAR_SSL_CLIENT_OBUF_SIZE
#define BEAR_SSL_CLIENT_OBUF_SIZE 2048 + 85
#endif

#ifndef BEAR_SSL_CLIENT_IBUF_SIZE
#define BEAR_SSL_CLIENT_IBUF_SIZE 32768 + 85 + 325 - BEAR_SSL_CLIENT_OBUF_SIZE
#endif
```

raleortiz, 2021-09-18:
> Problema resuelto mediante el uso de ArduinoBearSSL, pero necesita aumentar los búferes de entrada y salida de ArduinoBearSSL
> 
> Mis búferes son:
> 
> ```
> #ifndef BEAR_SSL_CLIENT_OBUF_SIZE
> #define BEAR_SSL_CLIENT_OBUF_SIZE 2048 + 85
> #endif
> 
> #ifndef BEAR_SSL_CLIENT_IBUF_SIZE
> #define BEAR_SSL_CLIENT_IBUF_SIZE 32768 + 85 + 325 - BEAR_SSL_CLIENT_OBUF_SIZE
> #endif
> ```

Do you have code to test?

asus1305, 2022-01-17:
@raleortiz @SMauser have you solve this problem? i have the same problem. 
@d0ct0rd-dalimo can you give us code to test? thank you