witnessmenow/Universal-Arduino-Telegram-Bot #243: Hello i got this error but idk how to fix can you help me please ?

MelihOzbk, 2021-06-05:
```
#include <deneyap.h>
#include <ArduinoJson.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

// Bağlanacağınız WiFi bilgileri. 
#define WIFI_SSID ""
#define WIFI_PASSWORD ""

// Telegram BOT bilgileri. 
#define BOT_TOKEN "" // Telegram uygulaması üzerinden Botfather ile öğreneceğiz. Burayı değiştirmeyi unutmayın.

const unsigned long BOT_MTBS = 1000; 

WiFiClientSecure secured_client;
UniversalTelegramBot bot(BOT_TOKEN, secured_client);
unsigned long bot_lasttime; 

#define ledPin D4 // Cihazın bağlı olacağı pin.
int ledStatus = 0;

void handleNewMessages(int numNewMessages)
{
  Serial.print("Cihaz ile iletişim kuruldu. ");
  Serial.println(numNewMessages);

  for (int i = 0; i < numNewMessages; i++)
  {
    String chat_id = bot.messages[i].chat_id;
    String text = bot.messages[i].text;

    String from_name = bot.messages[i].from_name;
    if (from_name == "")
      from_name = "Guest";

    if (text == "/ac")
    {
      digitalWrite(ledPin, HIGH); 
      ledStatus = 1;
      bot.sendMessage(chat_id, "Cihaz açıldı.", "");
    }

    if (text == "/kapat")
    {
      ledStatus = 0;
      digitalWrite(ledPin, LOW); 
      bot.sendMessage(chat_id, "Cihaz kapandı.", "");
    }

    if (text == "/durum")
    {
      if (ledStatus)
      {
        bot.sendMessage(chat_id, "Cihaz açık.", "");
      }
      else
      {
        bot.sendMessage(chat_id, "Cihaz kapalı.", "");
      }
    }

    if (text == "/start")
    {
      String welcome = "Merhabalar, DENEYAP Kart uzaktan röle kontrolü uygulamasına hoş geldin, " + from_name + ".\n";
      welcome += "Kullanabileceğin komutlar.\n\n";
      welcome += "/ac : Ledi yakar.\n";
      welcome += "/kapat : Ledi söndürür.\n";
      welcome += "/durum : Ledin durumu hakkında bilgi verir.\n";
      bot.sendMessage(chat_id, welcome, "");
    }
  }
}

void setup()
{
  Serial.begin(115200);
  Serial.println();

  pinMode(ledPin, OUTPUT); 
  delay(10);
  digitalWrite(ledPin, HIGH); 

 
  Serial.print("İnternete bağlanıyor: ");
  Serial.print(WIFI_SSID);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    secured_client.setCACert(TELEGRAM_CERTIFICATE_ROOT); 
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.print("\nWiFi bağlandı. IP adresi: ");
  Serial.println(WiFi.localIP());

  Serial.print("Alma süresi: ");
  configTime(0, 0, "pool.ntp.org"); 
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
  Serial.println(now);
}

void loop()
{
  if (millis() - bot_lasttime > BOT_MTBS)
  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);

    while (numNewMessages)
    {
      Serial.println("Dönüş alındı.");
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }

    bot_lasttime = millis();
  }
}

```

**Error;**
```
Arduino:1.8.15 (Windows 10), Kart:"Deneyap Kart, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, Auto, 921600, None"

C:\Users\EvPc\Documents\Arduino\sketch_jun05a\sketch_jun05a.ino: In function 'void setup()':

sketch_jun05a:90:28: error: 'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

   secured_client.setCACert(TELEGRAM_CERTIFICATE_ROOT); 

                            ^

"WiFi.h" için birden fazla library bulundu

Kullanılıyor: C:\Program Files (x86)\Arduino\libraries\WiFi

Kullanılmıyor: C:\Users\EvPc\AppData\Local\Arduino15\packages\deneyap\hardware\esp32\1.0.1\libraries\08_WiFi

exit status 1

'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

MelihOzbk, 2021-06-15:
got fixed lol

closed 2021-06-15 by MelihOzbk

trinican2, 2022-01-12:
I am getting the same error.  What did you do to resolve it ?

MelihOzbk, 2022-01-23:
i change ArduinoJson version to 5.x.x

MelihOzbk, 2022-01-23:
i saw error on arduino ide, i moved the library files to sketch folder and i changed the arduinojson version to 5.x.x. now, i am using platformio and i was able to run it without doing any of the things I said