witnessmenow/Universal-Arduino-Telegram-Bot #245: Not receiving response from EchoBot (Wifinina)

ykpo2021, 2021-06-09:
Hi all,

I am a beginner to Arduino world. When I upload EchoBot to my Arduino Board, it shows that "Connected to wifi". But afterwards, when I try to send out a message through telegram bot, serial monitor did not send out "get response":

     17:40:52.959 -> Attempting to connect to SSID: *****
     17:41:08.314 -> Connected to wifi
     17:41:08.314 -> SSID: *****
     17:41:08.348 -> IP Address: *****
     17:41:08.382 -> signal strength (RSSI):-54 dBm

Below are some of the information of board and libraries I am using:
     Arduino Board: Arduino Uno WiFi rev2
     WifiNINA.h: version 1.8.3
     ArduinoJson.h: version 6.18.0
     Arduino IDE: version 1.8.13

Can anyone please help? Appreciate so much for your hands.

Best regards,
PO

ykpo2021, 2021-06-10:
Hi all,

I tried to downgrade all the libraries used:
WifiNINA.h: version 1.0.0
ArduinoJson.h: version 5.13.5

This time, the serial monitor print out "got response", and a few second later, the telegram bot send out a message, which the content is same as the message I send out yesterday.
After that, I tried to send out message to telegram bot, but no response for both serial monitor and telegram bot.

I tried to modified the loop() as follow, to check the value of bot.getUpdates:
`void loop() {
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    Serial.print("Checking...");
    Serial.println(bot.getUpdates(bot.last_message_received + 1));
    while (numNewMessages) {
      Serial.println("got response");
      for (int i = 0; i < numNewMessages; i++) {
        bot.sendMessage(bot.messages[i].chat_id, bot.messages[i].text, "");
      }ot
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }

    Bot_lasttime = millis();
  }`

Result shows that the bot.getUpdates(bot.last_message_received + 1) is 0 at all time:
16:21:05.093 -> signal strength (RSSI):-50 dBm
16:21:09.977 -> Checking...0
16:21:14.813 -> got response
16:21:32.593 -> Checking...0
16:21:49.210 -> Checking...0

Please let me know if I miss something to seek for help. Thank you very much!

BR,
PO

Blon2000, 2021-07-23:
I've resolved same isssues with the Arduino MKR1010, installing the Nina Firmware 1.3.0 , instead the new rev 1.4.5

vale47, 2021-11-17:
Hello everyone, I am also a beginner into the arduino world, I've just started to use the new arduino 1 wifi rev 2, and I've experience the same problem of ykpo2021 with the Telegram Bot library. The wifi connection works properly while the bot.getUpdates does not receive any message. Did someone have solved this issue with this board? I don't know if a downgrade of the system could solve this problem, any clues?

dostick, 2023-01-23:
Hello,
I've tried to make this library's "Echobot.ino" example work with an Arduino Uno Wifi Rev2 with no success for more than two weeks :(

I've tired:
Different firmware versions
Different WiFiNINA.h versions
Different Arduinojson.h versions

Nothing seems to work. Any help would be appreciated.
Last test was using latest WiFiNINA.h version and ArduinoJson.h: version 5.13.5

Here's the code I'm using:

/*******************************************************************
 *  An example of bot that echos back any messages received        *
 *  using A101TelegramBot.                                          *
 *                                                                 *
 *  written by Giacarlo Bacchio (Gianbacchio on Github)            *
 *  adapted by Brian Lough                                         *
 *******************************************************************/

 #include <SPI.h>
 #include <WiFiNINA.h>
 #include <UniversalTelegramBot.h>

// Initialize Wifi connection to the router
char ssid[] = "";                // your network SSID (name)
char password[] = "";                              // your network key

// Initialize Telegram BOT
#define BOTtoken ""  // your Bot Token (Get from Botfather)

WiFiSSLClient client;
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

void setup() {

  Serial.begin(115200);

  delay(100);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  while (WiFi.begin(ssid, password) != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);

}

void loop() {

  if (millis() > Bot_lasttime + Bot_mtbs)  {
    Serial.println("Checking");
    Serial.println(bot.getUpdates(bot.last_message_received));
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    while(numNewMessages) {
      Serial.println("got response");
      for(int i=0; i<numNewMessages; i++) {
        bot.sendMessage(bot.messages[i].chat_id, bot.messages[i].text, "");
      }
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }
    Bot_lasttime = millis();
  }
}