witnessmenow/Universal-Arduino-Telegram-Bot #247: store chat_id into EEPROM

konig87nikkinakki, 2021-06-22:
Good morning,
thank you for your library and sorry i'm a hobbist and then some Examples are quite complicated for me. here's my issue:
1) i see in examples that BOT answers to the Sender using the chat_id , otherwise he cannot answer
2) i d like to NodeMCU sends me text even if i didn't write him, something when a door is opened.
3) so i obtained my ID of my telegram account, that is 7615439xx.
QUESTIONS:
1) can i use my chat_id into the script and send directly messages to my smartphone?
2) if i didnt know my ID, i can have it from your Library because there is
 "String chat_id = bot.messages[i].chat_id;"
so i can read it first time i write to the bot, the NodeMCU can store my id first time in EEPROM.
BUT chat_id you defined as STRING, i need it to be a Long Int, so what can i do to convert to it, store in eeprom and then read FROM EEPROM everytime it boot after power fail and convert again to string?

To avoid it, can your library send message to telegram and not to chat_id so everyone that are using the BOT can receive mexage automatically? something like "send to the chat" and not "send to the specified user"?
Thank you so much,
waiting for your help and sorry again for my inexperience, i'm a hobbist but not a beginner

witnessmenow, 2021-06-22:
>    4. can i use my chat_id into the script and send directly messages to
>    5. if i didnt know my ID, i can have it from your Library because

Hello

There is an example for the esp32 that does at least most of what your
looking for, you will have to port it to the esp8266 if you want to run it
on a nodemcu though

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP32/SendMessageFromEvent/SendMessageFromEvent.ino

Chat IDs are technically strings from telegrams AP so I won't be changing
the type, you can attempt to convert it yourself if you need it as a long
int. Unless there is a reason to that you need to use eeprom I would just
keep the chat Id in your sketch. I would also consider using littleFS
rather than eeporom if you need to store things on an esp8266 that persists
after reboots.

Sending messages to everybody: not possible, not supported by the telegram
api. If you want to send to everyone connected to the bot you need to keep
track of all the users and send them the message 1 by 1

konig87nikkinakki, 2021-06-22:
thank you for fast reply, really!
i ve seen the ESP32 example and could really work fine for me :-) Questions:
#define CHAT_ID "175753388"
so what is CHAT_ID ? string? int? does it work if i try with
#const long int = 175753388;
or 
long int = 175753388;
Everytime i program, i try to avoid to use #define because i really dont know how-arduino-see-it ;-) thanks

witnessmenow, 2021-06-28:
> #define CHAT_ID "175753388"
> so what is CHAT_ID ? string? int? does it work if i try with
> #const long int = 175753388;
> or
> long int = 175753388;

https://www.cplusplus.com/doc/tutorial/preprocessor/

> Everytime i program, i try to avoid to use #define because i really dont know how-arduino-see-it ;-) thanks
> 
> EDIT: if i use this method i have to know chat_id before to compile program. if i could read from a received message, i could store in eeprom and in next reboot i can read from eeprom and answer to the last sender of mex. i dont know if i explained :-D

Look up how to use spiffs or littleFs. 

Not an issue with the library

closed 2021-06-28 by witnessmenow