witnessmenow/Universal-Arduino-Telegram-Bot #248: Universal-Arduino-TelegramBot not send message

heliorcastro, 2021-06-26:
I have an ESP32 card working using Telegram and for over a year without problems. The board worked by receiving messages from the cell phone and also being able to send messages to the cell phone. About a week ago 6/16/21 I noticed that the card was no longer receiving a message from the cell phone, but it can send a message. I used another ESP32 board reprogrammed and nothing on the board received a message. I noticed by the ESP32 serial if the cell id reaches the ESP32 but it doesn't, I also created another Token and nothing. I suspect that Telegram has made some platform change that makes the UniversalTelegramBot lib not fully functional.

Roberto6969, 2021-06-27:
If this will help you - my ESP works with Telegram without any problem. Even more:  this days I have found out the ESP32 sent about 80 messages and it is still alive. This was not the case before, as it crashed/restarted after 25 messages max.

heliorcastro, 2021-06-27:
My ESP32 board keep sending messages it just suddenly stopped reveiving messages from the cell phone. I noticed some information that the Telegram is using https, situation not used in this lib UniversalTelegramBot.

witnessmenow, 2021-06-28:
What version of the library are you using?

I tested the latest version of the library this week on both esp32 and esp8266 and both worked fine

heliorcastro, 2021-06-29:
Hello Mr. Brian!

I used the library version 1.1.0 for a year that worked very well but about two weeks the sending of messages from cell phone to ESP does not occurs but the message is sent from ESP to cell phone.
Today I installed the library version 1.3.0 but it is showing an error when doing the compilation.
I simplified the code to find the cause of the error, I left only the basic library to analyze the error.
With version 1.1.0 there is no error in the compilation.
Below is the error that happens with version 1.3.0

#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <TelegramCertificate.h>
#include <UniversalTelegramBot.h>

void setup ()
{  
}
void loop ()
{  
}

Arduino: 1.8.13 (Windows 10), Board:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"
In file included from C:\Users\heliocastro\Documents\Arduino\libraries\UniversalTelegramBot\src/UniversalTelegramBot.h:31:0,
                 from E:\versoes_em _uso\TelegramBot_send_reception\TelegramBot_send_reception.ino:6:
C:\Users\heliocastro\Documents\Arduino\libraries\UniversalTelegramBot\src/TelegramCertificate.h:9:38: error: redefinition of 'const char TELEGRAM_CERTIFICATE_ROOT []'
 const char TELEGRAM_CERTIFICATE_ROOT[] = R"=EOF=(
                                      ^
In file included from E:\versoes_em _uso\TelegramBot_send_reception\TelegramBot_send_reception.ino:5:0:
C:\Users\heliocastro\Documents\Arduino\libraries\UniversalTelegramBot\src/TelegramCertificate.h:9:12: note: 'const char TELEGRAM_CERTIFICATE_ROOT [1369]' previously defined here
 const char TELEGRAM_CERTIFICATE_ROOT[] = R"=EOF=(
            ^
Used: C:\Users\heliocastro\Documents\Arduino\libraries\ArduinoJso
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1

witnessmenow, 2021-06-29:
You need to update Arduino JSON to V6 too.

It seems something happened recently to older versions of the library ok,
but I won't be investigating as it's working on the latest version

heliorcastro, 2021-06-30:
Mr. Brian!

I updated the lib of ArduinoJson from 6.10.0 to 6.18.0, I also needed to remove the preferences.txt file until the moment the compiler no longer indicated errors.
Now the message is sent from the cell phone to the ESP32 board and vice versa.
To work it was necessary to use UniversalTelegramBot version 1.3.0 and ArduinoJson 6.18.0.
Thank you for your full attention for pointing the way to this solution.

frostfire1, 2022-12-20:
i using universalTelegramBot version 1.3.0 and ArduinoJson 6.19.4. but still not work
`d:\Joe\Latihan\Arduino\libraries\Universal-Arduino-Telegram-Bot-1.3.0\src/TelegramCertificate.h:9:38: error: redefinition of 'const char TELEGRAM_CERTIFICATE_ROOT []'
 const char TELEGRAM_CERTIFICATE_ROOT[] = R"=EOF=(
                                      ^
In file included from D:\Joe\Tugas\Kir\Kresna IoT\Project\Project.ino:2:
d:\Joe\Latihan\Arduino\libraries\Universal-Arduino-Telegram-Bot-1.3.0\src/TelegramCertificate.h:9:12: note: 'const char TELEGRAM_CERTIFICATE_ROOT [1369]' previously defined here
 const char TELEGRAM_CERTIFICATE_ROOT[] = R"=EOF=(`

frostfire1, 2022-12-20:
> i using universalTelegramBot version 1.3.0 and ArduinoJson 6.19.4. but still not work
>
> d:\Joe\Latihan\Arduino\libraries\Universal-Arduino-Telegram-Bot-1.3.0\src/TelegramCertificate.h:9:38: error: [...]

Sorry my bad, when I looked at the "UniversalTelegramBot.h" file, it already included "TelegramCertificate.h"