witnessmenow/Universal-Arduino-Telegram-Bot #249: Echo bot ESP8266 - every second request seems to be falling

witnessmenow, 2021-06-28:
Every second request seems to be receiving an empty response:

(I've revoked this token)

```
sending: bot245123272:AAEOrezpTXZ4Bw52OBwWOLDNBHpRMActep0/getUpdates?offset=1&limit=1

{"ok":true,"result":[]}

incoming message length 23
Creating DynamicJsonBuffer
GetUpdates parsed jsonObj: {"ok":true,"result":[]}
no new messages
Closing client
time took (ms): 1608
GET Update Messages
[BOT]Connecting to server
sending: bot245123272:AAEOrezpTXZ4Bw52OBwWOLDNBHpRMActep0/getUpdates?offset=1&limit=1
Received empty string in response!
Closing client
time took (ms): 3021
GET Update Messages
[BOT]Connecting to server
sending: bot245123272:AAEOrezpTXZ4Bw52OBwWOLDNBHpRMActep0/getUpdates?offset=1&limit=1

{"ok":true,"result":[]}

incoming message length 23
Creating DynamicJsonBuffer
GetUpdates parsed jsonObj: {"ok":true,"result":[]}
no new messages
Closing client
time took (ms): 1541
GET Update Messages
[BOT]Connecting to server
sending: bot245123272:AAEOrezpTXZ4Bw52OBwWOLDNBHpRMActep0/getUpdates?offset=1&limit=1
Received empty string in response!
Closing client
time took (ms): 2711
GET Update Messages

```

solcer, 2021-06-28:
I am looking at the debug output of the code. The output is at below.  sendGetToTelegram function does not return with json string. 

Btw i am using 1.1.0 version.

[BOT]Connecting to server
.... connected to server

<!DOCTYPE html>
<html class="">
  <head>
    <meta charset="utf-8">
    <title>getUpdates</title>
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta property="description" content="">
    <meta property="og:title" content="getUpdates">
    <meta property="og:image" content="">
    <meta property="og:description" content="">
    <link rel="shortcut icon" href="/favicon.ico?4" type="image/x-icon" />

    <link href="/css/bootstrap.min.css?3" rel="stylesheet">
    
    <link href="/css/telegram.css?212" rel="stylesheet" media="screen">
    <style>
    </style>
  </head>
  <body class="preload">
    <div class="dev_page_wrap">
      <div class="dev_page_head navbar navbar-static-top navbar-tg">
        <div class="navbar-inner">
          <div class="container clearfix">
            <ul class="nav navbar-nav navbar-right hidden-xs"><li class="navbar-twitter"><a href="https://twitter.com/telegram" target="_blank" data-track="Follow/Twitter" onclick="trackDlClick(this, event)"><i class="icon icon-twitter"></i><span> Twitter</span></a></li></ul>
            <ul class="nav navbar-nav">
              <li><a href="//telegram.org/">Home</a></li>
<li class="hidden-xs"><a href="//telegram.org/faq">FAQ</a></li>
<li class="hidden-xs"><a href="//telegram.org/app

incoming message length1300
Creating DynamicJsonBuffer
Failed to parse update, the message could be too big for the buffer
GET Update Messages

philicibine, 2022-01-12:
I am also seeing this behaviour.

```
GET Update Messages
[BOT]Connecting to server
sending: bot1728637842:AAFLcUCaZGM3OwrPz3gDkxheGNeL0BH5ktk/getUpdates?offset=550743279&limit=1

{"ok":true,"result":[]}

incoming message length 23
Creating DynamicJsonBuffer
GetUpdates parsed jsonObj: {"ok":true,"result":[]}
no new messages
Closing client
GET Update Messages
[BOT]Connecting to server
sending: bot1728637842:AAFLcUCaZGM3OwrPz3gDkxheGNeL0BH5ktk/getUpdates?offset=550743279&limit=1
Received empty string in response!
Closing client
GET Update Messages
[BOT]Connecting to server
sending: bot--MYBOTTOKEN--/getUpdates?offset=550743279&limit=1

{"ok":true,"result":[]}

incoming message length 23
Creating DynamicJsonBuffer
GetUpdates parsed jsonObj: {"ok":true,"result":[]}
no new messages
Closing client
GET Update Messages
[BOT]Connecting to server
sending: bot--MYBOTTOKEN--/getUpdates?offset=550743279&limit=1
Received empty string in response!
Closing client
```