witnessmenow/Universal-Arduino-Telegram-Bot #255: +- 5 sec delay every time my code checks telegram

prontodos, 2021-09-19:
I am using D1 mini ESP8266

code snip
========
void loop() {
  currentMillis = millis();
  checktele();
  dosomething();
}

void checktele() {
  if (currentMillis - previousMillis >= interval) {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    while (numNewMessages) {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);

    previousMillis = currentMillis;
    }
  }
}

void dosomething() {
  Serial.println("dd");
}

Result form the console
==================
17:16:43.710 -> dd
17:16:47.354 -> dd
17:16:50.765 -> dd
17:16:54.210 -> dd
17:16:57.588 -> dd
17:17:01.002 -> dd
17:17:04.413 -> dd
17:17:07.826 -> dd
17:17:11.239 -> dd
17:17:14.651 -> dd
17:17:18.065 -> dd

As you can see from above my dosomething() is looping very slow about 5 seconds.

Why ? Anybody has encountered this ?
how to resolve this ?

witnessmenow, 2021-09-19:
I responded on the group to this, nearly all your delays are about 3.5 seconds, which is longer than I would expect, but not crazy off. Https requests on Arduino are just slow, I would expect around 2 seconds for an esp8266

Things to try:

- make sure your wifi network is strong where you are testing, esp boards do not have strong antennas
- there is an menu option in the Arduino ide for setting the clock speed of the esp8266 to 160Mhz , might be worth a try.

closed 2021-09-19 by witnessmenow

prontodos, 2021-09-20:
Got it thank you !!