witnessmenow/Universal-Arduino-Telegram-Bot #262: It not really issue

harutokagami, 2021-11-16:
is there a way to add more then 1 chat id currently I'm using `String CHAT_ID = "chat id here";`

petergeher, 2021-12-15:
this is how I did it. Not pretty, but does what it should. 

    if (chat_id != 0123456789) {
      if (chat_id != 1234567890) {
        if (chat_id != 2345678901) {
          if (chat_id != 3456789012) {
            if (chat_id != 4567890123) {
              …
              bot.sendMessage(chat_id, „nope!“, „“);
              continue;
            }
          }
        }
      }
          …
    }

tglaria, 2022-05-30:
> this is how I did it. Not pretty, but does what it should.
>
> ```
> if (chat_id != 0123456789) {
>   if (chat_id != 1234567890) {
>     if (chat_id != 2345678901) {
>       if (chat_id != 3456789012) [...]

You might want to try with switch() case?

Esque123, 2022-08-30:
You should be able to put the chat_id's into an array and then just use a for loop to loop through the array sending to "nope!" message to each chat id as it goes through each iteration of the loop.