witnessmenow/Universal-Arduino-Telegram-Bot #265: Bot token as variable from spiffs, possible?

petergeher, 2021-11-30:
Is it possible to take the bot token as a variable from the spiffs?
In principle I can read the token and also display it e.g. as 
Serial.print(spiffs_token); 
display it. But

UniversalTelegramBot(BOT_TOKEN, client).sendMessage(chat_id, welcome, "");

does not work like this.  Anyone have any idea how to do this? Would hate to have the token hard in the code.

witnessmenow, 2021-11-30:
Yeah it is possible. 

- Set up a bot like the examples (even with a blank token : "")
- Use `updateToken("bot12356")` to update the token whenever you need to https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.h#L71

petergeher, 2021-12-02:
Thank you very much, this works perfectly! 
Now to solve my favorite problem; 'if else etc' for more than one user. But I will manage that :-) 

    // ChatID test
    if (chat_id != Peter) {
      bot.sendMessage(chat_id, "Nope!", "");
      continue;
    }
    // ChatID test

Best regards,
Peter

closed 2021-12-02 by petergeher