witnessmenow/Universal-Arduino-Telegram-Bot #266: anonym message-"bomb" kills bot

umbm, 2021-12-02:
For testing I sent a large text to the bot and the bot doesnt work anymore.

I activated #define TELEGRAM_DEBUG 1 in UniversalTelegramBot.h and got this:

> incoming message length 1500
> Creating DynamicJsonBuffer
> Failed to parse update, the message could be too big for the buffer. Error code: IncompleteInput
> Closing client
> GET Update Messages
> [BOT]Connecting to server
> sending: bot2 . . . [cut]

This runs in an infinite loop and the bot does not work anymore.

Biggest problem: 
I sent the too large message from an not connected account to the bot (before /start command) and this runs in this issue.
Means: everybody who finds the bot can kill it with a simple too large text-message.

umbm, 2021-12-02:
Suggestion to solve the problem:
Get the update_id from message what creates the error:
`{"ok":true,"result":[{"update_id":115492776,`

Start a new request with offset update_id **+1** (at least) 
`/getUpdates?offset=115492777&limit=1`

This marks the message as read and the bot is running again.

If someone gets in this trap meanwhile you can solve it manually going 3 steps:

Get the next message of the bot in browser:
`https://api.telegram.org/bot[your bot token here]/getUpdates?offset=1&limit=1`
Take the update_id from response, add 1 and call bot again with the offset.
`https://api.telegram.org/bot[your bot token here]/getUpdates?offset=115492777&limit=1`

RobertGnz, 2022-01-12:
I ran into the same problem. See #275 here [https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/275](url)
The problem comes from a bug in UniversalTelegramBot::getUpdates() . I suggested a patch to automatically solve the problem.

francwalter, 2022-09-29:
Done.
Created [my own fork](https://github.com/francwalter/Universal-Arduino-Telegram-Bot)
And I created a [pull request with the fix from Robert](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/pull/296).  But very unlikely that this pull could get pulled once. The original Repo is dead since a while (anyway very useful and good work).
See also [the fork from Jose](https://github.com/J-Rios/Universal-Arduino-Telegram-Bot) where he "fixes" the use of Strings. But it is behind master of original, its more to get ideas where to change the strings.