witnessmenow/Universal-Arduino-Telegram-Bot #278: 'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

ProtonMeow, 2022-02-22:
Arduino: 1.8.20 Hourly Build 2021/12/20 07:33 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Pyton_to_Arduino:46:15: error: 'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

   46 | X509List cert(TELEGRAM_CERTIFICATE_ROOT);

      |               ^~~~~~~~~~~~~~~~~~~~~~~~~

Multiple libraries were found for "UniversalTelegramBot.h"

 Used: C:\Users\User\Documents\Arduino\libraries\UniversalTelegramBot

 Not used: C:\Users\User\Documents\Arduino\libraries\Universal-Arduino-Telegram-Bot-master

exit status 1

'TELEGRAM_CERTIFICATE_ROOT' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Loler-proger, 2022-08-02:
have you found the solution?

Bolukan, 2022-10-23:
This would suggest having installed multiple versions of this library.
Check the file UniversalTelegramBot.h in those maps whether around line 31 you see `#include <TelegramCertificate.h>`.
If not update the library first.

ProtonMeow, 2022-10-25:
i already found the solution, actually its has been a long time since i fix this problem. i believe you need to download and install universal telegram from the sketch>library in Arduino IDE if it's doesn't work, update all your library, if it's still can't work than restart your Laptop. i hope its help! >_<

closed 2022-10-25 by ProtonMeow