witnessmenow/Universal-Arduino-Telegram-Bot #283: Understanding the "String Parse Mode"? WemosD1Mini, C++, MQTT project

BruceCrowthorne, 2022-03-27:
I am having problems finding the documentation on how to format data to include it in a message that I want to send.
My data comes from the MQTT messaging service and is in the form of an array of bytes. I want to put that message into a Telegram message and send it out.
I am having problems finding out how to format a message that the "Arduino-Telegram-Bot" will accept. I am starting simple by taking an existing string and adding a another string to it (using strcat) - but I can't even get that to work!

It works fine if the string is included in quotes --  bot.sendMessage(CHAT_ID, "Bot started up", "");
But not if I build a string and try to send the variable that is the string --

`  char secondmessage = "highmount-b16";   // declare variable & put value in
  strcat (thirdmessage, "test string");               // add a second string
  bot.sendMessage(CHAT_ID, thirdmessage, "");`     // try and publish it - FAIL!

As you can see I am leaving the String-Parse-Mode as an empty field. I assume that is my problem??

So I am looking for documentation on how to correctly format the message, can you point me in the right direction, please?

Just as an aside, I overcame one problem of the Telegram Bot and the MQTT library needing different Wifi clients - that was a bit tricky! If you would like the code I am happy to share it.

Thanks in advance
Bruce

BruceCrowthorne, 2022-03-27:
Sorry, the formatting of the code messed up, let me try again ...

`char secondmessage = "highmount-b16"; `
`strcat (thirdmessage, "test string");
`bot.sendMessage(CHAT_ID, thirdmessage, "");`

Maurizio1, 2022-04-16:
Hi BruceCrowthorne,
just have a look at this: [https://core.telegram.org/bots/api#formatting-options]