witnessmenow/Universal-Arduino-Telegram-Bot #285: SecureClient not working on espressif32@4.2.0

Darkest, 2022-05-04:
After upgrading platform to espressif32@4.2.0 from 3.5.0 in platform.io, SecureClient works badly, it successfully connects for the first time, even reads and posts some msg to telegram, but after it is closed, it can never be opened once again with the error connect(): start_ssl_client: -1

dgtaheno, 2022-05-11:
Hello,

Also experiencing the same issue here.

Best regards

witnessmenow, 2022-05-13:
Maybe try change this to `if(true)` to see if that fixes the issue

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.cpp#L773

dgtaheno, 2022-05-17:
Hello  Brian,
Thank you for your fast feedback, unfortunately this is not fixing my problem, when trying to send the first message to Telegram, not even pictures I receive the following error:
"[WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1"
In infinite loop.
It is fixed if I downgrade spressif to 4.0.0.
If you need further context, the project I´m working in: [https://github.com/dgtaheno/Imail/](url)
Thank you in advance for your time.
Best regards.

zella, 2022-05-20:
4.3.0 - worked.

witnessmenow, 2022-05-21:
> 4.3.0 - fixed

Tested fixed? Must have been a bug with the 4.2 release so

zella, 2022-05-22:
Yes. I upgrde my project from 4.2 to 4.3 and error gone

SirSerow, 2022-08-02:
Hi everyone! 

Just updated my espressif32 to 5.1.0 and got the same problem. Any thoughts on why this even may occur and how to fix it manually?

Thanks in advance.