witnessmenow/Universal-Arduino-Telegram-Bot #290: I cannot communicate with the bot using an Arduino Nano 33 iot

TheMechatronicEngineer, 2022-07-04:
I've tried for the first time the example echobot using an arduino nano 33 iot and the classical arduino ide. I have created a bot using botfather. I have uploaded the code of the example on the board. The serial monitor tells me that it's connected to the wifi and shows me the SSID the IP address and the signal strenght but when I try to write something using the telegram bot, nothing happens: I don't receive the echo message in the bot chat. Can anybody help me?

```
// ----------------------------
// Standard Libraries
// ----------------------------
#include <SPI.h>

// ----------------------------
// Additional Libraries - each one of these will need to be installed.
// ----------------------------
#include <WiFiNINA.h>
// Library for using network deatures of the official Arudino
// Wifi Boards (MKR WiFi 1010, Nano 33 IOT etc)

// Search for "nina" in the Arduino Library Manager
// https://github.com/arduino-libraries/WiFiNINA

#include <UniversalTelegramBot.h>
// Library for connecting to Telegram

// Search for "Telegram" in the Arduino Library Manager
// Install the "Universal Telegram" one by Brian Lough
// https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

#include <ArduinoJson.h>
// Library used for parsing Json from the API responses

// Search for "Arduino Json" in the Arduino Library manager
// https://github.com/bblanchon/ArduinoJson

// Wifi network station credentials
char ssid[] = "SSID";         // your network SSID (name)
char password[] = "password"; // your network password
// Telegram BOT Token (Get from Botfather)
#define BOT_TOKEN "XXXXXXXXX:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

const unsigned long BOT_MTBS = 1000; // mean time between scan messages

int status = WL_IDLE_STATUS;

WiFiSSLClient client;
UniversalTelegramBot bot(BOT_TOKEN, client);
unsigned long bot_lasttime; // last time messages' scan has been done

void handleNewMessages(int numNewMessages)
{
  for (int i = 0; i < numNewMessages; i++)
  {
    bot.sendMessage(bot.messages[i].chat_id, bot.messages[i].text, "");
  }
}

void printWiFiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

void setup()
{
    //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if (fv < "1.0.0") {
    Serial.println("Please upgrade the firmware");
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, password);

    // wait 10 seconds for connection:
    delay(10000);
  }
  Serial.println("Connected to wifi");
  printWiFiStatus();
}

void loop()
{
  if (millis() - bot_lasttime > BOT_MTBS)
  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);

    while (numNewMessages)
    {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }

    bot_lasttime = millis();
  }
}
```

tadcrazio, 2022-10-30:
Did you figure it out? I'm having a similar issue with my RP2040. It looks like because we are not setting the ssl cert that telegram API won't work. I don't see a way in Wifi101 or Wifinina to set a cert or fingerprint.

gmcode85, 2023-05-10:
I have the same issue using a Arduino Wifi, with echobot.

I have turned the debug for this library on and I can see it joining the server okay and can send, but cannot receive anything.

witnessmenow, 2023-05-10:
> 

You could maybe try a staticjson buffer instead of a dynamic one in the get updates method, I think esp devices have more dynamic memory than most, I'm not sure though

gmcode85, 2023-05-11:
Hi,

Thanks for getting back to me. A log is below. The message "Boogy" is 
out of date and it seems to be stuck on this message. The rest of the 
data is correct.

10:57:05.512 -> Attempting to connect to SSID: ZZZZ
10:57:18.056 -> Connected to wifi
10:57:18.056 -> SSID: ZZZZ
10:57:18.088 -> IP Address: 192.168.1.190
10:57:18.120 -> signal strength (RSSI):-62 dBm
10:57:18.151 -> [BOT]Connecting to server
10:57:19.458 -> sending: bot5653269344:ZZZ/getMe
10:57:19.635 ->
10:57:19.635 -> 
{"ok":true,"result":{"id":56532693,"is_bot":true,"first_name":"ZZZZ","username":"ZZZZ","can_join_groups":true,"can_read_all_group_messages":false,"supports_inline_queries":false}}
10:57:19.833 ->
10:57:19.833 -> Closing client
10:57:20.768 -> sendPostMessage: SEND Post Message: 
{"chat_id":"1599143","text":"Bot started up"}
10:57:20.864 -> [BOT Client]Connecting to server
10:57:22.314 -> Posting:{"chat_id":"1599143","text":"Bot started up"}
10:57:22.429 ->
10:57:22.429 -> 
{"ok":true,"result":{"message_id":510,"from":{"id":56532693,"is_bot":true,"first_name":"Graham","username":"ZZZZ"},"chat":{"id":1599143,"first_name":"ZZZZ","type":"private"},"date":1683795442,"text":"Bot 
started up"}}
10:57:22.685 ->
10:57:22.685 -> 
{"ok":true,"result":{"message_id":510,"from":{"id":56532693,"is_bot":true,"first_name":"Graham","username":"ZZZZ"},"chat":{"id":1599143,"first_name":"ZZZZ","type":"private"},"date":1683795442,"text":"Bot 
started up"}}
10:57:22.909 -> Closing client
10:57:22.942 -> GET Update Messages
10:57:22.983 -> [BOT]Connecting to server
10:57:24.221 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:24.507 ->
10:57:24.507 -> {"ok":true,"result":[{"update_id":129698933,
10:57:24.570 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:24.796 ->
10:57:24.796 -> incoming message length 256
10:57:24.828 -> Creating DynamicJsonBuffer
10:57:24.859 -> GetUpdates parsed jsonObj: 
{"ok":true,"result":[{"update_id":129698933,"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:25.148 ->
10:57:25.148 -> resultArrayLength:1
10:57:25.180 -> processResult: updateId:0 last_message_received:0 
messageIndex:0
10:57:25.245 ->
10:57:25.245 -> return newMessageIndex0
10:57:25.277 -> numNewMessages: 0
10:57:26.202 -> GET Update Messages
10:57:26.234 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:26.453 ->
10:57:26.453 -> {"ok":true,"result":[{"update_id":129698933,
10:57:26.485 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:26.709 ->
10:57:26.709 -> incoming message length 256
10:57:26.742 -> Creating DynamicJsonBuffer
10:57:26.773 -> GetUpdates parsed jsonObj: 
{"ok":true,"result":[{"update_id":129698933,"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:27.062 ->
10:57:27.062 -> resultArrayLength:1
10:57:27.093 -> processResult: updateId:0 last_message_received:0 
messageIndex:0
10:57:27.158 ->
10:57:27.158 -> return newMessageIndex0
10:57:27.189 -> numNewMessages: 0
10:57:28.146 -> GET Update Messages
10:57:28.178 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:28.523 ->
10:57:28.523 -> {"ok":true,"result":[{"update_id":129698933,
10:57:28.555 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}

-------- Original Message --------

  		SUBJECT:
  		Re: [witnessmenow/Universal-Arduino-Telegram-Bot] I cannot 
communicate with the bot using an Arduino Nano 33 iot (Issue #290)

  		DATE:
  		2023-05-10 16:13

  		FROM:
  		Brian Lough ***@***.***>

  		TO:
  		witnessmenow/Universal-Arduino-Telegram-Bot 
***@***.***>

You could maybe try a staticjson buffer instead of a dynamic one in the 
get updates method, I think esp devices have more dynamic memory than 
most, I'm not sure though

gmcode85, 2023-05-11:
`10:57:05.512 -> Attempting to connect to SSID: ZZZZ
10:57:18.056 -> Connected to wifi
10:57:18.056 -> SSID: ZZZZ
10:57:18.088 -> IP Address: 192.168.1.190
10:57:18.120 -> signal strength (RSSI):-62 dBm
10:57:18.151 -> [BOT]Connecting to server
10:57:19.458 -> sending: bot5653269344:ZZZ/getMe
10:57:19.635 ->
10:57:19.635 -> 
{"ok":true,"result":{"id":56532693,"is_bot":true,"first_name":"ZZZZ","username":"ZZZZ","can_join_groups":true,"can_read_all_group_messages":false,"supports_inline_queries":false}}
10:57:19.833 ->
10:57:19.833 -> Closing client
10:57:20.768 -> sendPostMessage: SEND Post Message: 
{"chat_id":"1599143","text":"Bot started up"}
10:57:20.864 -> [BOT Client]Connecting to server
10:57:22.314 -> Posting:{"chat_id":"1599143","text":"Bot started up"}
10:57:22.429 ->
10:57:22.429 -> 
{"ok":true,"result":{"message_id":510,"from":{"id":56532693,"is_bot":true,"first_name":"Graham","username":"ZZZZ"},"chat":{"id":1599143,"first_name":"ZZZZ","type":"private"},"date":1683795442,"text":"Bot 
started up"}}
10:57:22.685 ->
10:57:22.685 -> 
{"ok":true,"result":{"message_id":510,"from":{"id":56532693,"is_bot":true,"first_name":"Graham","username":"ZZZZ"},"chat":{"id":1599143,"first_name":"ZZZZ","type":"private"},"date":1683795442,"text":"Bot 
started up"}}
10:57:22.909 -> Closing client
10:57:22.942 -> GET Update Messages
10:57:22.983 -> [BOT]Connecting to server
10:57:24.221 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:24.507 ->
10:57:24.507 -> {"ok":true,"result":[{"update_id":129698933,
10:57:24.570 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:24.796 ->
10:57:24.796 -> incoming message length 256
10:57:24.828 -> Creating DynamicJsonBuffer
10:57:24.859 -> GetUpdates parsed jsonObj: 
{"ok":true,"result":[{"update_id":129698933,"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:25.148 ->
10:57:25.148 -> resultArrayLength:1
10:57:25.180 -> processResult: updateId:0 last_message_received:0 
messageIndex:0
10:57:25.245 ->
10:57:25.245 -> return newMessageIndex0
10:57:25.277 -> numNewMessages: 0
10:57:26.202 -> GET Update Messages
10:57:26.234 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:26.453 ->
10:57:26.453 -> {"ok":true,"result":[{"update_id":129698933,
10:57:26.485 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:26.709 ->
10:57:26.709 -> incoming message length 256
10:57:26.742 -> Creating DynamicJsonBuffer
10:57:26.773 -> GetUpdates parsed jsonObj: 
{"ok":true,"result":[{"update_id":129698933,"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:27.062 ->
10:57:27.062 -> resultArrayLength:1
10:57:27.093 -> processResult: updateId:0 last_message_received:0 
messageIndex:0
10:57:27.158 ->
10:57:27.158 -> return newMessageIndex0
10:57:27.189 -> numNewMessages: 0
10:57:28.146 -> GET Update Messages
10:57:28.178 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:28.523 ->
10:57:28.523 -> {"ok":true,"result":[{"update_id":129698933,
10:57:28.555 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}`

gmcode85, 2023-05-11:
10:57:05.512 -> Attempting to connect to SSID: ZZZZ
10:57:18.056 -> Connected to wifi
10:57:18.056 -> SSID: ZZZZ
10:57:18.088 -> IP Address: 192.168.1.190
10:57:18.120 -> signal strength (RSSI):-62 dBm
10:57:18.151 -> [BOT]Connecting to server
10:57:19.458 -> sending: bot5653269344:ZZZ/getMe
10:57:19.635 ->
10:57:19.635 -> 
{"ok":true,"result":{"id":56532693,"is_bot":true,"first_name":"ZZZZ","username":"ZZZZ","can_join_groups":true,"can_read_all_group_messages":false,"supports_inline_queries":false}}
10:57:19.833 ->
10:57:19.833 -> Closing client
10:57:20.768 -> sendPostMessage: SEND Post Message: 
{"chat_id":"1599143","text":"Bot started up"}
10:57:20.864 -> [BOT Client]Connecting to server
10:57:22.314 -> Posting:{"chat_id":"1599143","text":"Bot started up"}
10:57:22.429 ->
10:57:22.429 -> 
{"ok":true,"result":{"message_id":510,"from":{"id":56532693,"is_bot":true,"first_name":"Graham","username":"ZZZZ"},"chat":{"id":1599143,"first_name":"ZZZZ","type":"private"},"date":1683795442,"text":"Bot 
started up"}}
10:57:22.685 ->
10:57:22.685 -> 
{"ok":true,"result":{"message_id":510,"from":{"id":56532693,"is_bot":true,"first_name":"Graham","username":"ZZZZ"},"chat":{"id":1599143,"first_name":"ZZZZ","type":"private"},"date":1683795442,"text":"Bot 
started up"}}
10:57:22.909 -> Closing client
10:57:22.942 -> GET Update Messages
10:57:22.983 -> [BOT]Connecting to server
10:57:24.221 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:24.507 ->
10:57:24.507 -> {"ok":true,"result":[{"update_id":129698933,
10:57:24.570 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:24.796 ->
10:57:24.796 -> incoming message length 256
10:57:24.828 -> Creating DynamicJsonBuffer
10:57:24.859 -> GetUpdates parsed jsonObj: 
{"ok":true,"result":[{"update_id":129698933,"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:25.148 ->
10:57:25.148 -> resultArrayLength:1
10:57:25.180 -> processResult: updateId:0 last_message_received:0 
messageIndex:0
10:57:25.245 ->
10:57:25.245 -> return newMessageIndex0
10:57:25.277 -> numNewMessages: 0
10:57:26.202 -> GET Update Messages
10:57:26.234 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:26.453 ->
10:57:26.453 -> {"ok":true,"result":[{"update_id":129698933,
10:57:26.485 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:26.709 ->
10:57:26.709 -> incoming message length 256
10:57:26.742 -> Creating DynamicJsonBuffer
10:57:26.773 -> GetUpdates parsed jsonObj: 
{"ok":true,"result":[{"update_id":129698933,"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}
10:57:27.062 ->
10:57:27.062 -> resultArrayLength:1
10:57:27.093 -> processResult: updateId:0 last_message_received:0 
messageIndex:0
10:57:27.158 ->
10:57:27.158 -> return newMessageIndex0
10:57:27.189 -> numNewMessages: 0
10:57:28.146 -> GET Update Messages
10:57:28.178 -> sending: bot5653269344:ZZZ/getUpdates?offset=1&limit=1
10:57:28.523 ->
10:57:28.523 -> {"ok":true,"result":[{"update_id":129698933,
10:57:28.555 -> 
"message":{"message_id":493,"from":{"id":1599143,"is_bot":false,"first_name":"Graham","language_code":"en"},"chat":{"id":1599143,"first_name":"Graham","type":"private"},"date":1683711745,"text":"Boogy"}}]}

gmcode85, 2023-05-11:
It´s failing on update_id always being 0

  // Check have we already dealt with this message (this shouldn't happen!)
  if (last_message_received != update_id) 
  {

I might be incorrect assumption

gmcode85, 2023-05-12:
Hi,

After half a day of debugging and learning about json objects etc, I tracked it down to the following fails on an Uno because an int is only 2 bytes, not 4.

bool UniversalTelegramBot::processResult(JsonObject result, int messageIndex) {
  int update_id = result["update_id"];

Make it a long the same as last_message_received that it compared to and it works.

long update_id = result["update_id"];

garagesteve1155, 2023-11-05:
Anyone ever figure this out? Im having the same issue with my 33 IOT

tadcrazio, 2023-11-13:
> Anyone ever figure this out? Im having the same issue with my 33 IOT

FWIW I had switched to a different Telegram client, believe it was AsyncTelegram2 and didn't have issues there.

SimoneMSR, 2024-08-31:
I have the same issues reported in this closed [topic](https://forum.arduino.cc/t/universal-arduino-telegram-bot-doesnt-work/1009283/10).

Basically, neither with UniversalTelegramBot, nor with AsyncTelegram2 I was able to receive or send messages by using the libraries examples.

I solved after reading this [example](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/echoBot_wifiNINA/echoBot_wifiNINA.ino) from github.

I just needed to add telegram certificate to arduino. That is:
```
http://api.telegram.org
```