witnessmenow/Universal-Arduino-Telegram-Bot #302: Including emoticons

henkjannl, 2022-10-20:
This is not an issue but a contribution: I want to share the way how to use emoticons in bots. Previously I used unicode characters to produce emoticons but that did not always work well.

I'm using the EchoBot example, but modified the ```handleNewMessages``` function:

```c++
void handleNewMessages(int numNewMessages)
{
  String response = "";
  String newChar = "";
  for (int i = 0; i < numNewMessages; i++)
  {
    for(int j = 0; j<bot.messages[i].text.length(); j++) {
      newChar = String( (int) bot.messages[i].text[j],HEX);
      newChar.toUpperCase();
      response += String("\\x") + newChar;
    }
    response+= "\n"+bot.messages[i].text;

    bot.sendMessage(bot.messages[i].chat_id, response, "");
  }
}
```

If you now send an emoticon to the test bot, the bot will respond with the hexadecimal characters that represent the emoticon.

You can add the emoticons in your code as follows:

```c++
const String EMOTICON_BULLSEYE           = "\xF0\x9F\x8E\xAF";
const String EMOTICON_GLASSES            = "\xF0\x9F\x91\x93";
const String EMOTICON_STHETOSCOPE        = "\xF0\x9F\xA9\xBA";
const String EMOTICON_SHOWER             = "\xF0\x9F\x9A\xBF";
const String EMOTICON_BATH               = "\xF0\x9F\x9B\x81";
const String EMOTICON_FLAME              = "\xF0\x9F\x94\xA5";
const String EMOTICON_POINTING_FINGER    = "\xF0\x9F\x91\x89";
const String EMOTICON_THERMOMETER        = "\xF0\x9F\x8C\xA1";
const String EMOTICON_UP_ARROW           = "\xE2\xAC\x86\xEF\xB8\x8F";
const String EMOTICON_DOWN_ARROW         = "\xE2\xAC\x87\xEF\xB8\x8F";
const String EMOTICON_TRIANGLE_UP        = "\xF0\x9F\x94\xBC";
const String EMOTICON_TRIANGLE_DOWN      = "\xF0\x9F\x94\xBD";
const String EMOTICON_RED_QUESTION_MARK  = "\xE2\x9D\x93";
const String EMOTICON_PIN                = "\xF0\x9F\x93\x8D";
const String EMOTICON_OFFICE             = "\xF0\x9F\x8F\xA2";
const String EMOTICON_HOUSE              = "\xF0\x9F\x8F\xA0";
const String EMOTICON_ISLAND             = "\xF0\x9F\x8F\x96";
const String EMOTICON_CALENDAR           = "\xF0\x9F\x93\x85";
const String EMOTICON_CHECKMARK          = "\xE2\x9C\x85";
const String EMOTICON_CROSSMARK          = "\xE2\x9D\x8C";
const String EMOTICON_LAMP               = "\xF0\x9F\x92\xA1";
const String EMOTICON_WARNING            = "\xE2\x9A\xA0\xEF\xB8\x8F";
const String EMOTICON_CLOCK              = "\xE2\x9A\xA0\xEF\xB8\x8F";
const String EMOTICON_FOOTSTEPS          = "\xF0\x9F\x91\xA3";
const String EMOTICON_RUNNER             = "\xF0\x9F\x8F\x83";
const String EMOTICON_MAGIC_STICK        = "\xF0\x9F\xAA\x84";
const String EMOTICON_BED                = "\xF0\x9F\x9B\x8F";
const String EMOTICON_CHECKERED_FLAG     = "\xF0\x9F\x8F\x81";
const String EMOTICON_STOPWATCH          = "\xE2\x8F\xB1";
const String EMOTICON_CLOCK              = "\xE2\x8F\xB0";
const String EMOTICON_HOURGLASS          = "\xE2\x8C\x9B\xEF\xB8\x8F";
```
In any message from a bot, you can incorporate one of these constants in the string to reply with an emoticon.

Hope this is useful.

witnessmenow, 2023-04-01:
This is kind of cool!

Maybe it could just be a separate h file attached to the library, so people could include it if they wanted, and just not if they didn't

henkjannl, 2023-04-04:
Hi Brian, thanks, please add it to the library if you please. Perhaps add the code as additional example so users can extend the list if they want to. HenkJan