witnessmenow/Universal-Arduino-Telegram-Bot #307: bot-Not-Sending-Messages-On-Event

hwguyturnsw, 2022-11-14:
Using ESP32-WROOM-32 Module (WEMOS-LOLIN-32 and custom board) with Arduino IDE. Bot will not send messages to me.

Versions:
Arduino IDE: 1.8.19
ESP32 Board Package: 2.0.5
ArduinoJSON: 6.19.4
Universal-Telegram-Bot: 1.3.0

I created the bot, started it, and checked the ChatID and API Token about a hundred times just to make sure.

Using ;;;bot.sendMessage(CHAT_ID, "Bot started up", "");;;; to send messages.

Could not get the messages from bot. Can anyone help? I tried downgrading versions of the libraries to see if that would help and I lost track on what I had done. Updated all libraries back to latest and still nothing.

It's probably something I've done (or haven't) that's screwed up. Any help would be greatly appreciated.
Thanks,
ejc

RamonAlves1357, 2023-03-02:
Estou com problema parecido

buguno, 2024-04-03:
I have the same problem, if I use the handler I send the message, but if I don't use the handler I don't not send message.

## My code without handle messages
```c++
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

#define BOT_TOKEN "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"
#define CHAT_ID "XXXXXXXXXXXXX"

const char* SSID = "XXXXXXXXXXXXX";
const char* PASSWORD = "XXXXXXXXXXXXX";

X509List cert(TELEGRAM_CERTIFICATE_ROOT);
WiFiClientSecure secured_client;
UniversalTelegramBot bot(BOT_TOKEN, secured_client);

void setupWifi();
void conectWifi();

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  delay(2000);
  
  setupWifi();
}

void setupWifi() {
  Serial.println("------Conexao WI-FI------");
  Serial.print("Conectando-se na rede: ");
  Serial.println(SSID);
  Serial.println("Aguarde");

  conectWifi();
}

void conectWifi() {
  if (WiFi.status() == WL_CONNECTED) {
    digitalWrite(LED_BUILTIN, HIGH);
    return;
  }

  WiFi.begin(SSID, PASSWORD);
  secured_client.setTrustAnchors(&cert);  // Add root certificate for api.telegram.org

  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
  }

  Serial.println();
  Serial.print("Conectado com sucesso na rede ");
  Serial.println(SSID);
  Serial.print("IP obtido: ");
  Serial.println(WiFi.localIP());
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {
  Serial.print("Trying sending message to Telegram, ChatID: ");
  Serial.println(CHAT_ID);
  const bool message_bot_return = bot.sendMessage(CHAT_ID, "In loop", "");
  Serial.println(message_bot_return);
  bot.sendMessage(CHAT_ID, "In loop", "");
  Serial.println("Ending");
  delay(100000);
}

```