witnessmenow/Universal-Arduino-Telegram-Bot #310: Add multi User_ID

DrSaad77, 2022-11-24:
Hello, it is more a help requested than an issue.
Can somebody please let me know how to add another User_ID at line #36 ??
there:  #define CHAT_ID "XXXXXXXXXX"

I want to add more than 1 Telegram number/ User_ID, meaning, receive notification from Arduino to multiple devices.
Should I maybe repeat this line with the others ID ??

Thx guys

Esque123, 2023-01-12:
1. You can create a list with the user ID's. eg.
const String authorized_chat_ids[] = { "1651071221" };

2. Then you need to be able to get the dynamic size of the list, so that the bot can latter iterate through the ID's.
const int numberUsers = sizeof(authorized_chat_ids) / sizeof(authorized_chat_ids[0]);

3. Then create a check function to verify new messages from users against the list that you created in step 1.
bool isValidUser(String chat_id, String from_name)
{
    for (int i = 0; i < numberUsers; i++) {
        if (authorized_chat_ids[i] == (chat_id)) {
            Serial.println(from_name + " is a valid user");
            return true;
            break;
        }
    }
    Serial.println(from_name + " is NOT a valid User!");
    return false;
}

4. Lastly under the "void handleNewMessages()" function, send messages to all authotized users.

    for (int i = 0; i < numNewMessages; i++) {
        String chat_id = bot.messages[i].chat_id;
        String from_name = bot.messages[i].from_name;
        if (isValidUser(chat_id, from_name)) {

            String text = bot.messages[i].text;

Hope this helps!

DrSaad77, 2023-01-12:
> 3\. bool isValidUser(String chat_id, String from_name)
>     {
>     for (int i = 0; i < numberUsers; i++) {
>     if (authorized_chat_ids[i] == (chat_id)) {
>     Serial.println(from_name + " is a valid user");
>     return true;
>     break;
>     }
>     }
>     Serial.println(from_name + " is NOT a valid User!");
>     return false;
>     }

Thx you very much fror your support. 
I wish to make a bot on ESP8622 connected to main GRID Power and get notificated when it is available on more than 1 device through Telegram. I am a novice at coding or arduino script... Anyway, I tried to follow your help, but could not apply point 4, under the "void handleNewMessages()" function, could not find it.

This is what I came out as a script:

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>
#include <ArduinoJson.h>

// Wifi network station credentials
#define WIFI_SSID "MYWIFI"
#define WIFI_PASSWORD "xxxxxxxxxx"
// Telegram BOT Token (Get from Botfather)
#define BOT_TOKEN "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

// Use @myidbot (IDBot) to find out the chat ID of an individual or a group
// Also note that you need to click "start" on a bot before it can
// message you

const String authorized_chat_ids[] = { "11111111", "22222222" };

const int numberUsers = sizeof(authorized_chat_ids) / sizeof(authorized_chat_ids[0]);

bool isValidUser(String chat_id, String from_name)
{
for (int i = 0; i < numberUsers; i++) {
if (authorized_chat_ids[i] == (chat_id)) {
Serial.println(from_name + " is a valid user");
return true;
break;
}
}
Serial.println(from_name + " is NOT a valid User!");
return false;
}

X509List cert(TELEGRAM_CERTIFICATE_ROOT);
WiFiClientSecure secured_client;
UniversalTelegramBot bot(BOT_TOKEN, secured_client);

void setup() {
  Serial.begin(115200);
  Serial.println();

 // attempt to connect to Wifi network:
  Serial.print("Connecting to Wifi SSID ");
  Serial.print(WIFI_SSID);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  secured_client.setTrustAnchors(&cert); // Add root certificate for api.telegram.org
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.print("\nWiFi connected. IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("Retrieving time: ");
  configTime(0, 0, "pool.ntp.org"); // get UTC time via NTP
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
  Serial.println(now);

  bot.sendMessage(CHAT_ID, "Bot started up", "");
  delay(15000);
  bot.sendMessage(CHAT_ID, "GRID is available !", "");
}

void loop() {

}

I tried to compile the code without point 4, and get this error: 
"CHAT_ID' was not declared in this scope"

Can you please take a look at the script and correct stuff, to be able to use it with 2 user_ID, and this way , I would be able to replicate it for 3 or more user.
Thx you very much.

witnessmenow, 2023-04-01:
This is a user level thing

```
const String authorized_chat_ids[] = { "11111111", "22222222" };
const int num_chat_ids = 2;

...
...

bool sendToAllUsers(String message){
    bool allResponse = true;
    for(int i =0; i< num_chat_ids ; i++){
        allResponse = allResponse && bot.sendMessage(authorized_chat_ids[i], message, ""); //if one fails, all response will be false
    }
    return false;
}

```

closed 2023-04-01 by witnessmenow