witnessmenow/Universal-Arduino-Telegram-Bot #314: Sending photos (Echobot example)

henkbb36org, 2023-01-06:
I am trying the Echobot example in order to learn how to send photos from the ESP32-CAM using UniversalTelegramBot 1.3 and the latest Arduino IDE. After uploading the sketch it runs and I see response when sending the /start, /flash and /photo command using my bot. However, when sending the /photo command I see the response "Sending" and "Done!" in the serial monitor but I don't receive a photo. When I add a "Serial.println(fb->len);" to the sendImage function I can see there is data, so the photo capture function seems to work. Any ideas?

henkjannl, 2023-01-25:
Same here, I also tried the /ESP32/SendPhoto/PhotoFromSD/ example but I get the same result with an M5stack and SD card: 
- SD Card is working fine
- File is found on the SD card
- Uploading the bytes takes a while 
- Serial port reports success
- Text message before and after the photo is sent well, so channel is open

I tried the same thing with SPIFFS instead of SD with the same result. Tried small (<4kb) and large photo's, JPG and PNG, but no result. Help would be very much appreciated.

henkbb36org, 2023-01-26:
Hi Henk Jan, I couldn't solve and I have switched to using the AscynchTelegram library. It has a working example using a bot. With these two functions from the example I can send a photo using a file from SPIFFS that was captured as a screenshot:

bool checkPhoto( fs::FS &fs ) {
  File f_pic = fs.open( FILE_PHOTO );
  unsigned int pic_sz = f_pic.size();
  return ( pic_sz > 100 );
}

void capturePhotoSaveSpiffs( void ) {
  camera_fb_t * fb = NULL; // pointer
  bool ok = 0; // Boolean indicating if the picture has been taken correctly
  
  SPIFFS.remove(FILE_PHOTO);
  do {
    // Take a photo with the camera
    Serial.println("Taking a photo...");

    fb = esp_camera_fb_get();
    if (!fb) {
      Serial.println("Camera capture failed");
      return;
    }
    // Photo file name
    Serial.printf("Picture file name: %s\n", FILE_PHOTO);
    File file = SPIFFS.open(FILE_PHOTO, FILE_WRITE);

    // Insert the data in the photo file
    if (!file) {
      Serial.println("Failed to open file in writing mode");
    }
    else {
      file.write(fb->buf, fb->len); // payload (image), payload length
      Serial.print("The picture has been saved in ");
      Serial.print(FILE_PHOTO);
      Serial.print(" - Size: ");
      Serial.print(file.size());
      Serial.println(" bytes");
    }
    // Close the file
    file.close();
    esp_camera_fb_return(fb);

    // check if file has been correctly saved in SPIFFS
    ok = checkPhoto(SPIFFS);
  } while ( !ok );
}

henkjannl, 2023-01-26:
Wow thanks very much, I want to be able to send logfiles from my room thermostat to my computer over Telegram and I was kind of stuck since non of the examples worked. Thanks for the hint I will certainly check out this library.

maxpicklez, 2023-02-17:
I had similar issues using an ESP32-cam AITHINKER with the Universal Telegram Bot from within Arduino Studio. I downloaded the latest master and it still didn't work. I realized that I was missing : 

client.setCACert(TELEGRAM_CERTIFICATE_ROOT); 

after my WiFi.Begin(SSID, WPA2);

After adding that line everything worked like a charm for me. I love this library. :+1:

Stepyon, 2024-04-25:
I have the same issue, photo just doesn't arrive with any settings I tried to modify, nothing helped, text messages arrive without problem, images don't.

FarhanSakti, 2024-12-12:
> I had similar issues using an ESP32-cam AITHINKER with the Universal Telegram Bot from within Arduino Studio. I downloaded the latest master and it still didn't work. I realized that I was missing [...]

I have exactly same issues. But in my example code there is client_setCACert line and still no image sent to my telegram bot. I'm using ESP32 Wrover Cam.